betaflight/betaflight #266: Predictive RC signal smoothing to shorten smoothing delay

nppc, 2016-03-11:
I tried in Excel a "predicting smoothing" algorithm for RC command to have it less choppy. The benefit of this idea is to smooth a signal with minimal delay.
The idea is simple. We analyze last 2-3 RC samples and predict a new one for every PID calc loop iteration.
What do you think, does this smoothing algorithm has more cons or pros or not worth of it?
Here is one simulation for RC signal at 50hz and PID loop time at 2khz: 
[image]
Thank you for considering this. :)
Pavel

laurynas-karvelis, 2016-03-11:
Hi @nppc, the smoothing function itself is very interesting to me. Could you share some pseudo code for this?

borisbstyle, 2016-03-14:
@nppc That is already in place set rc_smoothing = on. You get the minimum possible delay with smoothed RC....up to 1 RX cycle.

But IMO it is totally unneeded as there is no bad behaviour with choppy RC command. Especially not since we introduced DELTA_ON_MEASUREMENT method for D term

nppc, 2016-03-14:
Ok, thank you Boris. That's a good explanation.
Pavel

closed 2016-03-14 by nppc