betaflight/betaflight #268: Scisky 32bit Brushed FC gyro drifting issue.

shyney7, 2016-03-11:
If I set the motor_pwm_rate to 32000 to prevent the motors to start spinning when the battery is plugged in. The gyro starts to get negative degrees in the roll axis although the quad is laying flat. I first thought it's a defective gyro but since I'm not the only one with this problem and the fact that it only happens if the motor_pwm_rate is set to higher values I'm sure it's a firmware issue.

I was using betaflight 2.1.6 and because some people mention to update to a newer firmware I updated to the latest 2.5.3.

Before I tiped any commands in the CLI, I calibrated the acc and gyro. Then I waited to see if there is any drift. And NOTHING! No single drift to the left or the right.

Then I typed 
- set motor_pwm_rate = 32000
and
- set rc_smoothing = on

Then I tested if there is any drift and it started again to get negative degrees in the roll axis

So I changed the motor_pwm_rate to 16000. And there was still a drift in the roll axis BUT it wasn't that fast as if I had it set to 32000.

So I then changed the pwm_rate to 8000. And what happened is weird, the drift was fast again as if I had it on 32000 ???

So I then changed it again to 16000 and disabled rc_smoothing and I think the drift is now even slower by disabling rc_smoothing instead of just set the pwm_rate to 16000, but that is just my subjective sensation and I cant confirm if it has to do something with rc_smooting because the drift is still occurring.

But what I can confirm is that it has definitely something to do with the motor_pwm_rate. So it's a cleanflight problem.

This issue was already mentioned in the cleanflight github but I couldn't find anything in the betaflight github
https://github.com/cleanflight/cleanflight/issues/1780

I hope you can help us boris.

here is the rcgroups thread about this FC: http://www.rcgroups.com/forums/showthread.php?t=2466286

borisbstyle, 2016-03-18:
Are you calibrating gyro?

shyney7, 2016-06-15:
As I mentioned above I calibrated the gyro but after that it starts to drift. 
I was hoping that this issue would be solved in future releases but it's still happening.
I've tried it with the newest 2.8.1 release and this issue is still occuring.
Here is a post from me on rcgroups about this and it seems to be a issue that is afecting many scisky users. [http://www.rcgroups.com/forums/showthread.php?t=2466286&page=82#post34226614](url)

RcSam, 2016-06-26:
Count Me In to this problem. Doing whatever I can but can't get it to fly normal. Please someone post as soon as possible if he/she found a solution to this

borisbstyle, 2016-07-03:
3.0 has some brushed fixes

closed 2016-07-03 by borisbstyle

DevstickAds, 2016-08-08:
This problem is still an issue in 3.0 for me the only solution i've found is to use cleanflight 1.9 alientwiif1 target.  ACC calibrates to 0 and does not drift.