betaflight/betaflight #269: Motor rattling when yawing below min_check after stick arming on 2.5.3

robertlacroix, 2016-03-11:
Re-opened #242 as the problem is still happening in 2.5.3:

This is kind of weird new behavior of 2.5.3 (2.4.1 didn't have that problem). After arming by stick and then yawing below min_check, for example to disarm, the motors rattle weird. Check out this video to see what I mean:

http://youtu.be/ytsPoGwaqOY

borisbstyle, 2016-03-17:
min_check has been lowered in 2.5.3.....make sure your stick is really below min_check by certain margin. If it is close to it and hitting min_check often you indeed can get rattling

robertlacroix, 2016-03-18:
Saw the 1040 change. I just checked and throttle is at 987. Can you try to reproduce?

borisbstyle, 2016-03-31:
Try 2.6 with better D filtering and yaw filtering

closed 2016-03-31 by borisbstyle