betaflight/betaflight #274: Futaba S.Bus2 Telemetry support

StarrrLiteNL, 2016-03-12:
i was wondering if there are any plans to implement Futaba S.Bus2 support so that Futaba receivers can have telemetry sent back from the Flight Controller?

StarrrLiteNL, 2016-03-12:
http://sbustelemetrysensors.blogspot.nl has some detailed info about the s.bus2 protocol including sample code to do custom telemetry.

borisbstyle, 2016-03-23:
Anybody who wants to add this is welcome!

GadgetFPV, 2016-05-24:
So anyone interesting in adding Futaba SBus2 support? a lot of us still use Futaba, it would be nice to include...

maybeyes, 2016-06-14:
Yeah! it would be great! Someone can do this?

eltron247, 2016-07-12:
If anyone is able to create a modular style library for this, please let me know. It would be great to have not only for here on cleanflight but also for a number of other repo's. If you're interested please let me know. I'd love to compensate someone for their time to get this going.

closed 2016-08-19 by borisbstyle

hydra, 2016-08-20:
@eltron247 we'd welcome an implementation of this for Cleanflight.

DomLuu, 2017-08-14:
Can't find other "requests" regarding SBUS2 telemetry (Futaba) - is this feature request still open?