betaflight/betaflight #275: Impossible to arm on naze32

jeremaier, 2016-03-12:
Hello, I won't arm my motors through my radio control but when I use the motors tab and I increase a little bit the cursor, ESCs make a beep and their red led light on (KISS 30A). It worked with cleanflight and in an ancient version of betaflight(Sorry, I don't remember which one it was) but not with the last one (2.5.3). 
In the modes tab, it doesn't go to green when I put the switch on the arming interval like seen on the screenshot.
[image]

I applied these commands in CLI :
set acc_hardware = NONE
set baro_hardware = NONE
set mag_hardware = NONE
set deadband = 4
set yaw_deadband = 4
set max_aux_channels = 6
set gyro_lpf = OFF

And here is my dump :
[backup 12.03.16.txt](https://github.com/borisbstyle/betaflight/files/170545/backup.12.03.16.txt)

I hope you can help me and sorry for my english :/
Thanks

BibasCZ, 2016-03-13:
Try to set a value set min_throttle = 1250 under 1040.

jeremaier, 2016-03-14:
I tried it but It doesn't work :(

CreeT6, 2016-03-15:
try replicating the issue after a reflash

jeremaier, 2016-03-15:
I tried to reflash it many times and it doesn't work anymore. I tried also to flash it with cleanflight, apply just modifications in order to be able to arm (and see that it works with cleanflight) and applied exactly the same modifications on a betaflight version but it does not work

borisbstyle, 2016-03-15:
Can you do status, tasks and version and post your new full dump

jeremaier, 2016-03-15:
# version
# BetaFlight/NAZE 2.5.3 Mar  6 2016 / 14:16:58 (0aac025)
# status
System Uptime: 55 seconds, Voltage: 157 * 0.1V (4S battery - OK), CPU:3%
CPU Clock=84MHz, GYRO=MPU6050
Cycle Time: 503, I2C Errors: 0, config size: 1384

# tasks
Task list:
0 - (SYSTEM) max: 12us, avg: 0us, rate: 9hz, total: 0ms
1 - (PID) max: 529us, avg: 256us, rate: 998hz, total: 36sec
- - (GYRO)  rate: 1996hz
4 - (SERIAL) max: 491us, avg: 2us, rate: 99hz, total: 27ms
5 - (BEEPER) max: 12us, avg: 0us, rate: 99hz, total: 9ms
6 - (BATTERY) max: 40068us, avg: 1us, rate: 49hz, total: 65ms
7 - (RX) max: 71us, avg: 39us, rate: 49hz, total: 194ms

The only thing I change between the first dump and this one is the min_throttle from 1250 to 1030 
[dump.txt](https://github.com/borisbstyle/betaflight/files/174454/dump.txt)

Thanks

borisbstyle, 2016-03-18:
I see you are using PWM......not seeing many folks using PWM nowadays. Ask someone else to verify who uses PWM perhaps.

My tests only go as far as PPM and serial protocols :D

borisbstyle, 2016-03-18:
Do you see bars moving under rx tab?

champino, 2016-03-18:
i still use the pmw signal. 2.4.1 was my last update. anything later will put my cpu at 11% and not allow my quad to arm.

jeremaier, 2016-03-19:
I find what was the problem :D 
I try every last releases since the 2.4.1 and I saw that I was unable to arm from the 2.5.1 then I looked in the release note and here it is : Default min_check lowered to 1040 (make sure your radio can go below 1040 to be able to arm)
My radio, a futaba had a min throttle of 1077...
My suggestion is that you should mention it more than this because I didn't see this requirement anywhere :/ and I think that this problem does not happen only to me

Thanks borisbstyle

borisbstyle, 2016-03-19:
the original multiwii / baseflight / cleanflight manuals all describe to
configure your radio to 1000-2000 range :)

The reason it is changed is because min_check also determines the lowest
possible throttle position and with 1100 you lose 10% of lower throttle
when your radio is set to 1000-2000 like recommended

jeremaier, 2016-03-19:
Oh okay, I understand :)
Thanks

closed 2016-03-19 by jeremaier