betaflight/betaflight #276: Beeper should not warn for "Lost RX" until RX has actually been on (or copter armed, or...)

MasterZap, 2016-03-14:
For various testing, it gets quite annoying to hear the copter beep by just plugging in the battery, or plugging it into USB or similar.

Lost RX beeping only makes sense if  you had any RX to lose, or maybe only if you have flown.

My suggestion is to add a check if the copter was ever armed since boot, and only THEN apply the beep on RX lost.

/Z

borisbstyle, 2016-03-14:
there is a beeper command to enable/disable beeper commands.

I still think it is a valid security feature warning someone that his RX is not turned on if they want that.

MasterZap, 2016-03-14:
What command would you suggest to achieve what I want? 
I of course not do _not_ want to turn of TX beep *completely*, only turn it of when it is pointless?

phredieFPV, 2016-03-16:
I would love to see an option to disable the buzzer prior to arming as MasterZap suggested. It gets annoying to have to put a bit tape over the buzzer every time I want to connect over USB. But if there is already a way to achieve this please enlighten me.

universam1, 2016-03-16:
+1
Since BF is primarily a often changed OS hence a good part of its live
connected to computer without TX necessarily turned on, I would highly
appreciate to connect the logic to the failsafe detection which already
offers the logic of determinating of a previous flight.
Thanks

Highoctane32, 2016-03-16:
I too think this would be a good idea. I mean as a safety feature for powering up without the tx on..if something bad is going to happen the beeper won't stop it from happening, in my experience any issues like failsafe on the rx set with throttle ect happen so quick that  by the time one realizes the beeper is beeping because the tx is off it is too late..and if it's just to notify one that tx is off of they can't figure that out within a second of it not working perhaps they shouldn't be flying?
Am I missing something?

MasterZap, 2016-03-17:
I don't think so. Except maybe for the guy standing on the starting line of a race for $250,000 forgetting to turn is Tx on, this shouldn't be a problem. I really vote strongly for this feature.

borisbstyle, 2016-03-17:
You can just disable all beeps from cli :D

StarrrLiteNL, 2016-03-17:
I understand the question, but on the other hand, imagine the Flight Controller spontaneously rebooting in mid air (bug in the code, loose connection, brownout etc). I would still want my beeper to go off after the quad has gone down to make finding it easier.

Highoctane32, 2016-03-17:
If it reboots as long as you still have radio link you could 1) have beeper on a switch. 2) turn off tx and it would beep, since when it rebooted the rx as long as in range would reconnect to the tx satisfying the proposed only beep after rx connection to the tx first time. For what you are describing to cause the beeper to fail would have to be the perfect storm of inflight reboot coupled with simultaneous failsafe/ out if RC range. Which is probably far less likely than the battery coming unplugged in a crash :)

StarrrLiteNL, 2016-03-17:
I totally agree, I just wanted to give 1 possible example of why the requested functionality might not always be desirable :)

flounderscore, 2016-03-18:
I also vote to keep the lost signal beeper as it is. You can also disable the beeper from the CLI while debugging.

borisbstyle, 2016-03-18:
@StarrrLiteNL I have my beeper on switch to find it. When RX would come loose my ESC's will beep after 10 minutes anyway

borisbstyle, 2016-03-18:
beeper on switch is only I personally use......beep only whenever I want it to beep!

MasterZap, 2016-03-19:
I made a pull request for a fix for his...

Arakon, 2016-03-20:
Compiled this PR into BF yesterday, works like a charm for me.

MasterZap, 2016-03-21:
Boris merged my change, so I'm closing this. Yay. :)

closed 2016-03-21 by MasterZap