betaflight/betaflight #278: Horizon mode on MW23 PID controller behaves strangely

MasterZap, 2016-03-14:
Maybe it is just me, or my settings, put Horizon levelling when using MW23 PID controller is near nonexistant.

Yes, it does seem to level... after about 5 seconds it may be horizontal. Since the use cases* I have for horizon needs "instant stick reaction", this is not quite what I want.

Am I just setting it up wrong?

/Z

* = a few use cases: 

1.  "rescue" mode when I've acro:ed myself into confusion 
2.  help me land with very tilted camera 
3.  I actually preffer an angle based mode for "zig-zag" flying (i.e. weaving between trees), but it needs to react to stick movement instantly, I want zero mush, and no "acro mode feel" at all. Just the copter angle glued directly to the stick angle, no delay.

MasterZap, 2016-03-14:
Hmm. Reading the code I see that the D term is used as some kind of "P limiter"? Odd. I'll have to play with that.

borisbstyle, 2016-03-17:
I will close this issue as level defaults etc are still not determined for MW23

closed 2016-03-17 by borisbstyle

MasterZap, 2016-03-17:
ok, I'll try it out some more, but it is behaving.... oddly. I have slow wobbles like too much very-slow I going on...