betaflight/betaflight #280: Insane Idea: Replace TPA with accelerometer based PID adjustment

MasterZap, 2016-03-15:
TPA is the "classic" way to adjust PID's based on power applied. However, isn't acceleration measurement a better way to detect the same problematic state? Should we adjust PID's to acceleration vector magnitude?

I know, it is a mad idea,  but.... who knows? :)

elf128, 2016-03-16:
So how does it supposed to behave flying inverted during roll?

CreeT6, 2016-03-16:
i was thinking we can mix Power applied with Z acceleration and when there is power but no acc, it is in  the propwash? and reduce pid?

elf128, 2016-03-16:
It doesn't work this way.
Going down with 1G deceleration is not any different from going up with 1G acceleration. However, airflow on props is complettely different. On top of that, once you brake an assumption that gravity is always pointing toward negative Z, things become even more complicated.
I don't see any benifits. 
Height difference is way more informative for propwash detection.

MasterZap, 2016-03-16:
Think if it this way: The situations where I have most issues with propwash is in tight 180degree turns, i.e. in "high G manouvers". Hence, I was thinking.... we could detect the "high G manouver" case and somehow treat it differently. That is all.

I said it was an insane idea.... :)

borisbstyle, 2016-03-17:
TPA isn't really necessary with the new liniear mixer.

TPA was actually just a workaround to lack of low throttle stability due to PID contstraining on the old mixer.

Prop wash scenario should not require less PID's in most cases. Sometimes a voltage PID compensation can do wonders on the right setup. Most prop wash problems really are in the ESC's

MasterZap, 2016-03-17:
Ok, You are probably rigtht.

I told you it was an insane idea :)

closed 2016-03-17 by MasterZap