betaflight/betaflight #281: Throttle set less then 1100, but motors jump  instantly to 1600+

proggod, 2016-03-17:
When i barely give it any throttle, the motors go full speed.  Setting the power of each motor individually works fine, the receiver tab shows everything is fine as well.  Tried to log, not sure what else you need.

Capture of motor tab during it happening: http://prnt.sc/ag922s
Capture of receiver tab during it happening: http://prntscr.com/ag9278

This exact config works great on two other quads, using same flight controller.  Switching to cleanflight it works fine with same config as well.

Attaching a log file and a config dump

[betaflight_config.txt](https://github.com/borisbstyle/betaflight/files/177207/betaflight_config.txt)

proggod, 2016-03-17:
Here is the log, had to zip it so it would take it:

Highoctane32, 2016-03-17:
This sounds like normal iterm wind up when on the ground and above min check..motors will increase speed as the FC is trying to make adjustments but they aren't having any effect either because craft is on the ground or props are off. Question though..why do you have min command set to 885? Also what are the ESCs calibrated to?

proggod, 2016-03-17:
Esc is set to 1000-2000 calibrated..   Its not normal it jumps off the ground so fast, i can't even bring it down without it dropping like a rock.  Its either on or off.  I change min command to 885, because sometimes with oneshot motors twitch without it.   Basically its at full throttle the minute i give it any throttle, its either full throttle or nothing.

borisbstyle, 2016-03-17:
Sounds like ESC's are not properly calibrated. Check what the blhelisuite says

proggod, 2016-03-17:
They are definitely calibrated, two things :)   One the motor tab works fine, i can slow up the power, motors work as normal, secondly it works fine in clean flight.

borisbstyle, 2016-03-17:
@proggod 
I mean calibrated as in terms what the blhelisuite says. Some boards like rev6a can calibrate ESC's every time on boot after you actually calibrated it to false values

proggod, 2016-03-17:
Yes it does show calibrated, if I go into
Motor tab and pick 1100 it spins slower and if I do 1200 it goes quicker so on and so forth, so that's working fine.  Also remember same config and setup works fine in cleanflight so it's not a hardware or config issue from what I can tell

borisbstyle, 2016-03-17:
Are you testing in acc mode?

proggod, 2016-03-17:
I have tested in rate mode and horizon mode, does it in both, the minute i give it even the tiniest bit of throttle it launches off the ground like a rocket ship.

borisbstyle, 2016-03-17:
It really sounds like your min_throttle value is just too high.

but also too high pids can cause erratic behaviour

proggod, 2016-03-17:
Ya, well that is first thing i changed, had to set it to 1050 just to keep it from popping off ground.  These motors are pretty powerful so I am wondering if its just correcting too much for their power.   What is even weirder, is I just pulled the latest cleanflight from github and flashed that, and it has terrible problems.  Instead of launching straight up, now the motors seem unstable, and it flips every time.  

Maybe I should just get rid of these motors/esc combo.  I have never liked 100% how it flew even when it did work well with cleanflight.

proggod, 2016-03-17:
Raceflight developer said he added something that may fix an issue like this to the latest release.   Right now I am waiting for him to make me some sparky binaries, to see if that does fix it.   He hasn't updated git yet with the code.

Highoctane32, 2016-03-17:
What ESCs and ESC FW version? I see you mentioned race flight, are you using Multishot? I've seen guys have to lower min command to say 950 at the lowest to get the twitching to stop, but 885? Was it doing the same thing before you set min command to 885. And I see you said the ESCs show calibrated in BLHeli. What are the exact values they show, if you don't mind humoring me?

proggod, 2016-03-17:
Well i tried it at 1000 too, that really doesn't seem to be an issue.  They were the emax nano's with oneshot.   What is great is I HAVE FIXED IT :)

I reset everything back to stock, reflashed firmware.  Only change what i absolutely needed to get it flying.   I have a standard set of pids I have been using in all my quads, that have them flying great, but I think that could have been very well the problem.  Maybe because the motors are so much more powerful, they need a lot lower pids.  I don't know exactly what caused it, but it is fixed now..

closed 2016-03-17 by proggod

Highoctane32, 2016-03-17:
Glad you got it figured out. Who knows.. I started my EMax rs2205/2600 on stock Luxfloat  PIDs with no issues, so unless you were trying to run mw23 on defaults doubt that was it. With that said a full chip erase and set up manually is my FiRST troubleshooting step with wierd issues. Have fun flying.

proggod, 2016-03-18:
i wasn't running default pids, i was running the pids that have always worked for me on all my other quads, so maybe it was my f*ckup

PascalFPV, 2016-03-18:
Im having a problem with my quad. When i give around 15% throttle 2 of the 4 motors give very highh throttle for a very short time. Could someone help me?

I tried alot of different betaflight versions and nothing helps.
im using : 

naze32 rev6a
rs2205 2300kv
little bee 30a

Ben je nederlands, kan je me op whatsapp bereiken 0628220231

thanks

proggod, 2016-03-18:
Sounds like similar to my issue, did you calibrate the esc, are you running oneshot, and did you try default settings?

PascalFPV, 2016-03-18:
Yes i tried everything. I noticed that when airmode is enabled the it
oscilates way more. Now im flying acro and its running pretty good. But
prefer airmode

proggod, 2016-03-18:
If your actually flying, it sounds like a PID tuning issue, I couldn't even fly, mine just went up like a rocketship, no control.

TomTiki, 2017-05-16:
In case it's helpful to anyone, I had the exact same issue and I finally solved it by re-flashing the Betaflight firmware to the flight controller.   The FC had been working fine and then I changed receivers and had to reprogram the model in my transmitter.  I then went into Betaflight Configurator to check and adjust my mode and throttle settings, etc.  Suddenly the throttles were all wacky.  I'd move the stick just a little up from zero and all the motors would nearly instantaneously go to full or near-full throttle.  I messed around for a looooong time with the throttle parameters in Betaflight Configurator and in BLHeli Configurator but nothing helped.  Finally I had to learn how to re-flash the FC board and that did the trick.

Bestkillroy, 2017-06-14:
Hi guys I'm relatively new to the hobby I've built about 15 quads but only 5 or 6 Brushless so I'm still in the hardcore learning and problem solving phase but I'm having his same issue with a new build of mine, I'm running a furious fpv racewhoop with a 2S battery and emax 6a bullet blheli-s esc's and a 96mm crazepony X-Style frame, and I'm experiencing this same issue, running oneshot125 and ESCs seem to be calibrating fine and motors spin up beautifully in motors tab but when I test it, as soon as I give it enough throttle to recognize it goes full throttle and would cut a flip were I not holding it and holding it, it shakes violently back and forth, I've tried every esc protocol and tried lowering my min throttle which seemed like it changed it a bit for the better but certainly not enough to fly it. I'm at a loss here guys and any advice y'all may have would be OH SO APPRECIATED.

toff75, 2022-01-18:
i did this. the motors needed to be reversed dur...