betaflight/betaflight #284: Naze32 poor motor power with oneshot125 after update to v2.5.x

leppo5940, 2016-03-19:
### Problem
on my Naze32 with oneshot125 activated I have poor motor power at full throttle. without oneshot everything works fine. I recognized it after update from v2.4 to all versions > 2.5.0RC7.

### Setup 
Board: Naze32v6
Versions: Betaflight 2.5.0RC7 and 2.5.3

tests done with default settings

### things I have done
after power on the board and measure the motor output signal with an arduino I get
110 µs when the motor slider in the configurator is at 1000
165 µs at 1500
200 µs at 1860

when I make a reset (reboot) of the FC in the configurator I get right signals and full motor power
125 µs at 1000
188 µs at 1500
232 µs at 1860

when I do a powercycle I get the bad values from above until I make an reset (reboot) in the configurator.
With oneshot125 off everything works ok.

borisbstyle, 2016-03-19:
How are you actually measuring? Are you using the sliders im the
configurator or do you arm it and really measure the output?

Are you using PWM for your RX perhaps?

leppo5940, 2016-03-19:
I used the sliders in the configurator to measure the output.

But the issue is exact what the quad does in flight. 
When I put in the battery, arm and fly the quad I have not the full power than expected.
Then I plug in the USB or sometimes I take bluetooth connection and reset the quad without power
down it has full power.
And that is exact what the measurements say.

The ESC´s have latest BL-Heli with calibrated range 1060-1860. And when the max_throttle is 
200µs and not 232µs I don't have the full power.

It seems as if there is an negative offset in the output signal, which is fixed after a reboot (reset)
 until you perform a powercicle.

The issue is not I the older v2.4 or in the latest cleanflight

borisbstyle, 2016-03-19:
There is no negative offset. It may sound like a timer clash in some
scenarios.

Thats why I asked if you were using PWM or PPM for your rx?

I will measure a naze32 soon and see what it puts out.

BTW try old version with emf_avoidance ON. That is enabled by default on
the latest version for naze32

leppo5940, 2016-03-20:
with emf_avoidance = on on v2.4.0 i get wrong output timings.
118µs at 1000 
220µs at 1860
with emf_avoidance = off i have normal timings.
It doesn´t make any different if i make a reboot or not.

In v2.5.x it doesn´t make any different when i play with the rx inputs (PWM,Seriell,PPM..) or with V_Bat on off. The issue is still there.

with v2.5.0Rc7 after powering on i get this cpu clock and the wrong timings

version:
 BetaFlight/NAZE 2.5.0 Mar  2 2016 / 23:55:59 (27c2a5d)
status:
System Uptime: 69 seconds, Voltage: 0 * 0.1V (1S battery - NOT PRESENT), CPU:4%
CPU Clock=64MHz, GYRO=MPU6500, ACC=MPU6500, BARO=BMP280
Cycle Time: 1004, I2C Errors: 0, config size: 1376

then i make save(reboot) and after that i get
status
System Uptime: 9 seconds, Voltage: 0 * 0.1V (1S battery - NOT PRESENT), CPU:3%
CPU Clock=84MHz, GYRO=MPU6500, ACC=MPU6500, BARO=BMP280
Cycle Time: 1000, I2C Errors: 0, config size: 1376

and the timings are ok

borisbstyle, 2016-03-22:
@leppo5940 
It is normal that on older cleanflight/betaflight versions it causes shift in timing. However in 2.5 this has been downscaled with 7/6th to correct it.

84Mhz / 72Mhz = 7/6th

borisbstyle, 2016-03-23:
i was not able to replicate this issue.

Can you somehow describe the steps what you do to reproduce it?

leppo5940, 2016-03-24:
Hello Boris,

1. 
 I write this code to an arduino for measuring the PWM signal:

    byte PWM_PIN = 3;
 
    int pwm_value;
 
     void setup() {
      pinMode(PWM_PIN, INPUT);
      Serial.begin(115200);
    }
 
    void loop() {
      pwm_value = pulseIn(PWM_PIN, HIGH);
      Serial.println(pwm_value);
    }

Connect Pin D3 from the arduino to any Motor output of the FC

2.
flashing Betaflight with Cleanflightconfigurator with option: _full flash erase_

3. 
In the configurator under _SETUP_  I make _RESET SETTINGS_

4.
Enable _Motor Test Mode_ 
and measure the motor PWM output with the arduino

5.
entering the CLI Mode and  type _status_
`System Uptime: 476 seconds, Voltage: 0 * 0.1V (1S battery - NOT PRESENT), CPU:5%
CPU Clock=64MHz, GYRO=MPU6500, ACC=MPU6500, BARO=BMP280
Cycle Time: 1005, I2C Errors: 0, config size: 1376`

6.
type _save_ in CLI

7.
going back to CLI tab and type _status_ again:
`System Uptime: 33 seconds, Voltage: 0 * 0.1V (1S battery - NOT PRESENT), CPU:3%
CPU Clock=84MHz, GYRO=MPU6500, ACC=MPU6500, BARO=BMP280
Cycle Time: 1006, I2C Errors: 0, config size: 1376`

now the CPU Clock changed from 64MHz to 84MHz, and the measured PWM timings have the right value.
When you power down and power up the FC I get the wrong PWM timings again until i make a reboot with the configurator.
In real flight conditions the problem expresses in having no full power until I make a reboot of the FC with the configurator. 
v2.5.4 #257 has the same issue.

borisbstyle, 2016-03-24:
>    1. In the configurator under *SETUP* I make *RESET SETTINGS*

Wait....I see one thing what triggers me!

You say you are seeing 64Mhz!

That indicates that this board may be missing the oscillator or it is
faulty!

Naze32 board should have 72Mhz on pre 2.5 versions and 84mhz on 2.5

64Mhz is a fallback speed on the MPU for when oscillator is malfunctioning.

Do you have another naze board perhaps?

borisbstyle, 2016-03-24:
Going to close this issue as this is 100% hardware issue on thid board.

Motor timing is derived from the clockspeed. Your faulty oscillator / mpu reduces the speed of clock pulses what lowers the power of your motors

closed 2016-03-24 by borisbstyle

leppo5940, 2016-03-25:
Hi Boris, 
You'r right, ist was a faulty oscillator. I could measure the 12MHz with an Oszilloskop. So i think the Oszillator starts up too slow.
Without the Oszillator the FC runs always with 64MHz. With a new Oszillator it works. The FC starts with 84 MHz. 
So if somewone has the same issue, change the Oszillator. 
Thanks Boris for your help and for your great work with the Software

borisbstyle, 2016-03-25:
No problem! Great that it works now