betaflight/betaflight #288: New min_check default of 1040 is problematic

ethomas999, 2016-03-22:
The new 'min_check' default of 1040 (https://github.com/borisbstyle/betaflight/commit/65e17dd957889539014acdf403cb05210442d1f8) is problematic on my micro-quad setup (DX6i and RX_SERIAL receiver) because the lowest PWM value it can reach on most channels is about 1065 (even with endpoint/travel-adjust at maximum).  It's possible to get the throttle signal that low via the transmitter trim, but changing trim on other channels will mess up their center points.

The result is that stick commands that need a low PWM no longer work (i.e., stick disarm, trim commands via sticks).  I had to set 'min_check' back to 1100 to get things right again.  I think others will run into this and no know why it is happening.

--ET

lichtl, 2016-03-22:
Have you already tried rxrange? Set the 4 channels to min and max. This then would re-scale them to 1000-2000

ethomas999, 2016-03-22:
Using 'rxrange' to scale the values does resolve the issue.  I still think, though, that a number of users will be tripped up by the change in 'min_check'.

--ET

borisbstyle, 2016-03-22:
@ethomas999 I will raise it a little bit, but in general I don't see a point of having optimized parameters for worst case setups and radios.
The majority still has the recommended full 1000-2000 range on their channels.

ethomas999, 2016-03-22:
@borisbstyle Understood.  It'd be good if there was a receiver-calibration function in the configurator, like what is in the ArduPilot MissionPlanner.  Maybe someday...

--ET

closed 2016-03-22 by ethomas999