betaflight/betaflight #290: a lot of vibrations and wobbles!

raymonkroeze, 2016-03-22:
Hello,
Since a month or so i changed my 2300kv motors to 2700kv motors on my 180 quad. Since then i had a lot of vibrations i can't explain. At first i thougt i was caused by a too high of pgain. But i lowered the pgain as low until feels really out of place and still there are vibrations. I the meanwhile i tried the latest version and still the same problem. I've tried various of settings, i noticed if increase the dgain the vibrations will be less, but then i feel the quad feels not as locked in (my Dgain on pitch roll was about 35).
Last weekend i was out of HQ props and put my dal props on. And since then i have some really bad wobbles, (maybe caused by out of balanced props, i don't know).
Also a friend of mine has exactly the same issue on a 210 quad running 2205 2300kv motors.
My setup:
180 frame
2205 2700kv motors
littlebee 20A 14.3 blheli
naze32 on 2.6khz running PID 1
I also tried to run on 1khz again, but same issue.

please help!

borisbstyle, 2016-03-23:
You would have to post some logs to see what is wrong on your tune.

raymonkroeze, 2016-03-25:
I don't have acces to blackbox right now. Yesterday i've done some more testing. Since last weekend the quad is pretty much unflyable at 3 quarters of throttle. The quad is shaking as crazy. If i lower Pgain on roll and pitch increase Dgain it's a bit more flyable, but still shaking. And the week before is was only as i described as above. Now i believe one of the motors is bringing vibrations into the quad. The quad is a unibody 180 frame made of 4 mm carbonfiber, so the smallest vibrations are amplified a lot. 
Im trying to fix this and going to isolate the FC from the body.
But for some reason i think these motors are just to powerfull for this small quad.

raymonkroeze, 2016-03-25:
I've done more testing. Im pretty sure it's not a tuning issue but a hardware issue. for some reason there is some massive vibration in the frame. I can see a blur in the goggles. So you can close this issue!

borisbstyle, 2016-03-31:
Try 2.6 release and report back

closed 2016-03-31 by borisbstyle