betaflight/betaflight #293: False roll command being executed, not RX failsafe

jsweeney91, 2016-03-23:
[loggy.TXT](https://github.com/borisbstyle/betaflight/files/186503/loggy.TXT)
just after 1:20 of the blackbox recording is showing false commands being picked up from the
RX. I checked my failsafe wasnt the issue in this next log file:
[te.TXT](https://github.com/borisbstyle/betaflight/files/186508/te.TXT)
I armed and throttled up as usual and turned off the TX

naze32 rev6
betaflight failsafe is set to hold on all apart from the throttle

Issue previously raised and received a lot of attention:
https://github.com/borisbstyle/betaflight/issues/249

jsweeney91, 2016-03-23:
apologies the log file in question is 6/6 of loggy.txt

jsweeney91, 2016-03-23:
and 21 seconds into log file 4/6

borisbstyle, 2016-03-23:
Sorry but to me looks like a faulty radio....it commands full right roll.

You are the only one having this problem so far. None of the other reports is simillar to yours.

closed 2016-03-23 by borisbstyle

jsweeney91, 2016-03-24:
I am the not the only one experiencing this issue, if you check issue #249 there are people having the same problem:
**carebear482 commented 19 days ago**
I can confirm this issue with naze rev5 SBUS.
**dawnduran commented 19 days ago**
Confirm Issue with naze rev5 CPPM (D4R-II)
)
I'd agree that it wasn't a betaflight issue if I was experiencing it with cleanflight too. Also tried:
-new rx
-new fc
-new radio module
-updated radio firmware
The GUI on my radio isn't displaying any unusual commands but I'll try to capture the blackbox while I shake my radio to see if there's a loose connection or whatever

borisbstyle, 2016-03-24:
In the other cases it was the glitch caused by failsafe.
Thats where throttle goes to min_usec for a short while.

But in your case its roll stick going to full stick. That is not a scenario
being triggered by failsafe

jsweeney91, 2016-03-24:
how can you say theirs is caused by the failsafe? They are reporting the same issue, there wasn't any RCA done for them so saying it was failsafe is based purely on assumptions.

> 
**lefticals  commented 20 days ago  **
Same issue after a punchout when i cut throttle and roll seems like quad has a mind of its own for s plit second ... Either yaw or roll moves a few degrees

I sat last night for about half an hour giving RC input while shaking my radio(what a sight) while monitoring the receiver tab and the issue wasn't present. It's not the radio sending the command

borisbstyle, 2016-03-24:
in your case your roll stick goes full stick.

In case of lefticals quad moves just few degrees and not FULL STICK.
That really is just a tuning issue. In your case the actual roll stick
moves to the right max reflection....that input is coming from your rx.

RC-Slater, 2016-03-24:
Yours sounds like the momentary (less than one second) glitch that happens at the conclusion of a fast snap roll or flip when throttle is all the way down, am I right? 

If so, we've discussed it at length in the forum, and sort of decided that it is most likely caused by low throttle desync or prop stall.  One hypothesis is air resistance on prop prevents motor/ESC from responding as quickly as necessary at low throttle values.   Possible fix is to raise your min throttle.  Raise it significantly at first and see if issue goes away

RC-Slater, 2016-03-24:
Full disclosure, I'm on mobile and unable to look at your log

jsweeney91, 2016-03-24:
@RC-Slater the problem is that the FC is receiving a full roll right command, not that motors are stalling. In the event of a stall is there fake commands being sent?

borisbstyle, 2016-03-24:
@jsweeney91
exactly what I was saying.....all other reported things on other issue were
completaly different than yours.

If you try any possible firmware and all cause same full roll stick than I
would really start to suspect your radio.

Can you verify what happens when you generate failsafe? Just to make sure
your failsafe is not configured to full right stick

jsweeney91, 2016-03-24:
@borisbstyle It doesn't happen on other firmware but I'll re-try older firmware/cf today/tomorrow and get blackbox for you. 

RX failsafe disarms and kills the throttle, my arm switch is also my blackbox switch so it stops after failsafe engages. You can see that on the second log file on the ticket te.TXT

jsweeney91, 2016-03-25:
@borisbstyle just a quick note:
every time the twitch occurs the blackbox recording is showing a sharp increase in acceleration on motor 4 and a sharp decrease in motor 1 **before** the stick command is issued. See images below:
http://imgur.com/a/Io3pk
I'm guessing that the blackbox write queue is placing motor output before rc command as the timeframe is pretty small between them, but I thought it may be relevant

jsweeney91, 2016-03-26:
problem not present in most recent betaflight