betaflight/betaflight #295: If arm switch is on and you plug a battery in, motors spin up

fahrvergnuugen, 2016-03-25:
With 2.5.4, if my arm switch is ON (because it got bumped or I forgot to check it) and I plug a battery in, the motors will spin up. It was a hard lesson learned when I switched to betaflight so I could use airmode - before that I used motor_stop and armed with the stick.

But I just noticed that if it goes into fail safe, the motors stop and they will not rearm when failsafe is deactivated unless you disarm and re-arm by toggling the arm switch. The buzzer also changes pattern to tell you that its not in failsafe, but it won't arm.

So why not make it do the same thing the first time it powers on? Don't allow arming unless it was disarmed at least once. This would prevent some surprise paper cuts!

KiteAnton, 2016-03-25:
What if you got a glitch in the power supply and the board resets? Then you will not be able to recover as easy as it is now. For every patch like this that is introduced a new problem/consequence appears. 

I think it is easier if you set up your tx to be a bit more difficult to arm by mistake, e.g. using some sticky switch to activate and perhaps some warning sound if you have the arm switch in armed mode accidentally.

fahrvergnuugen, 2016-03-25:
I like the idea of using a sticky switch. I have a timer reset assigned to a spring loaded switch. I could setup a mix where i need to hold the timer reset in and toggle the arm at the same time in order for it to arm. 

But I can honestly say I have never had a board reboot mid flight, and if I did I most certainly crashed because my arming was done with the throttle low/yaw right position.

Yamaford, 2016-03-31:
Set small angle = 180 and you can arm any way you want. My x frame with the battery on the bottom is like this. OR acc= none does the same thing.

Yamaford, 2016-03-31:
fahrvergnuugen another tip is to set MIN check very low. That way if your throttle isn't dead bottomed out on the tx it will not arm.

KiteAnton, 2016-06-16:
Things like this can sometimes be easier to fix in the transmitter instead, that will reduce the complexity of the FW. I suggest you use a sticky switch on your transmitter (possible with a two switch activation) and then add a warning sound for when the switch are in wrong position without the sticky switch being active. This will prevent the use-case you mentioned above. 

This is an old issue so I close this, you (or anyone) are free to make a PR with proposed change if you find this really needed and then it can be discussed again.

closed 2016-06-16 by KiteAnton