betaflight/betaflight #296: Strange flips and crashes since 2.5.3

iconicj, 2016-03-25:
Hi Boris,

Appreciate all the work you do. In 2.5.3, after a successful flight and coming in to land in essentially a hover, the quad flipped upside down and went full throttle in to the ground. I figured it was a weird hardware failure like an ESC dying or something. The Zuul is out of commission still. Later,I flashed my zmr, tweaker, and qav with it. The tweaker and zmr began doing it as well, but I was at altitude in FPV and could recover. Flashed 2.5.4, same issue. Flashed back to 2.3.4 and the issues stopped. Flashed back to 2.5.4 and they resumed. 

Friend has messaged me saying the same thing. Mine were cc3d targets running 1khz, his was sp racing f3 running 2khz. I'm only using lux, he's tried it on lux and mw rewrite.

I watched it happen once in los, and it appeared as if it was commanded to rotate upside down. That is, once it began the rotation, I moved sticks in full direction to counter the rotation, but it couldn't be saved, and the rotation continued to a crash. It appeared to be a PID controller attempting to correct a non existent issue.

I know this doesn't give you much to go on. Hoping others can chime in on this one.

elf128, 2016-03-25:
Can you please make blackbox log. It would help a lot with investigations.
Another thing to concider. Can you confirm that FC didn't crash before your
drone have crashed. Can you arm after crash? Any blinking LEDs?

iconicj, 2016-03-25:
I haven't the ability to make a log now, with the damage incurred in the crashed I'm out of flyable quads.

In one instance, the quad was not damaged, and I simply flipped the quad upright, armed it, and flew the rest of the battery. During the time it was upside down, leds were normal indicating a disarmed quad.

Once moving forward, the issue does not occur - you can fly a entire battery pack, pushing 30 amps per ESC, flips, rolls, punch outs, no issues. Only coming in to land does it randomly flip and crash.

borisbstyle, 2016-03-25:
@iconicj 
I have seen simillar thing with faulty gyro or some kind of grounding issue

Highoctane32, 2016-03-25:
Only happens when coming in to land. Is this with airmode off or on? Kind of sounds like this maybe him/his craft(balance)having an issue with the non airmode lack of low throttle stability?  Just a suggestion.

iconicj, 2016-03-26:
My first thought was, oh well, hardware issue. Then it began doing it on all of my quads simultaneously. 

Airmode is off, at the point it occurs. I have my arm switch also setup for air mode, with disarm motors regardless of throttle off - so in landing first thing I do is pull that switch to disarm - air mode off. Soon as I land and kill the throttle, motors stop. Regardless, it occurs in the air, before touching the ground.

Just from watching the behavior of it, it looks like it's an issue with the PID controller either having a strange bug, or perhaps a limitation with cc3d and 2khz mode and lack of of cpu power. (It did happen on a Naze target.)

Highoctane32, 2016-03-26:
So it doesn't happen with airmode on? Still sounds like you are experiencing the lack of low throttle stability with airmode off..if so that is normal. I know there were versions that had more low throttle stability even without airmode but people complained of bouncy landings. So consider that perhaps before betaflight you probably either ran min throttle higher or used stick arming, in either case you probably never saw as low motor rpm as you do now..without airmode on this can cause instability at low throttle. Some set ups it may be worse than others depending on CG ect. Perhaps try not disabling airmode until right before touch down.

Highoctane32, 2016-03-26:
Should have said stick arming with motor stop :) back when that was the norm we never wee able to see motor rpm so low inflight.

iconicj, 2016-03-27:
Doesn't matter if air mode is on or off. Min throttle was 1075, dunno if that matters. Essentially, it does this at low throttle though.  Still doesn't look like a stability issue to me..looks like the fc is commanding a slow roll. It's kind I'd like an integral wind up on only one motor.

I flashed back to 2.1.6 - the problems went away.

iconicj, 2016-03-27:
Additional info .. After twelve packs today, this issue only occurs in air mode. Arming without am enabled never duplicated the issue. Air mode on resulted in 6 crashes in 3 packs

borisbstyle, 2016-03-28:
Sounds like ESC issue when the problem is gone in 2.1.6. In 2.1.6 full power was drastically being reduced. In later versions you get full power and your esc doesnt like that it seems

iconicj, 2016-03-29:
Please elaborate on that last statement.

Keeping in mind this is now happening on all 5 quads I use, using different motors and escs.
Additionally, this doesn't happen during fast flight. I've found if I just take off immediately with a punch out and land equally quickly it doesn't happen. It only happens around 20% throttle in a hover.

I did notice that it's more prone to do this on a properly tuned quad. My locked in quads do it often. Default pids, not so much.

Boappp, 2016-04-04:
I have a same problème on my vidéo (the motor stops at the front right):
https://www.youtube.com/watch?v=KJ_IfaOYHn4 (french issue http://forum.amr-asso.com/index.php?topic=2079.15)

the problem persists even in this test performed one by one :
_BF 2.6
_BF 2.5.4
_ESC firmware (Multishoot or Oneshoot)
_looptme (500 or 1000)
_ modes (angle, horizon, airmod, acro)
_ moving the position of the esc (there is always the motor stops at the front right)

but i tested a 2.1.6 (only on a single battery). no problèm. but I will continue to validate the test.
I analyzed all the test with the black box.

Sp racing bangood
DYS XM20A 
KINGKONG 6045bn
ZMR250 + Diatone

iconicj, 2016-04-04:
I must protest the faulty fear label. Nonetheless, after another week and a hundred more flights, this problem never occurred again after flashing to 2.1.6 the back to 2.5.4 - whatever happened seems to have corrected itself on all 5 quads. I have no explanation for this.

Boappp, 2016-04-04:
Ok retested a 2.1.6 -> no bug
and I flash on 2.6 (no wipe data), I test 2 battery -> no bug for now.

Boappp, 2016-04-05:
4 flights later, 
fifth flight one bung and crash,
sixth flight, no bug.......................................
I go activate Blackbox continuously

Boappp, 2016-04-05:
Here is the problem Blackbox File

https://www.dropbox.com/s/w9ga2z16ict5ps3/blackbox_log_2016-04-05_194256.TXT?dl=0

Open explore with cleanflight blackbox

The 7/9 flight
the transition to
1:12 fc sending a command to 100% on the front right engine because it does not react
1:19
1:23
1:25
1:27
1:30
1:31
1:32 (so abused)

Note that I've switched the motor and ESC is always the right front that bug. the problem comes from the FC, I'll try a new one.

iconicj, 2016-04-06:
I concur. Most of the crashes I've seen it do, it flips back and left, that would agree with the front right motor going 100% for no reason.

travmo, 2016-04-06:
It's interesting this is coming up as I had Joshua Bardwell do a Blackbox review video on this and both he and I suspected it was a loose prop. I had the same issue with a front motor going 100% during flips, for me it was the front left.

Boappp, 2016-04-06:
i order a new spracing 20€ and naze32rev5 10€ banggood for a test...

jsweeney91, 2016-04-06:
@iconicj behaviour matching my crash at 4:38:

https://www.youtube.com/watch?v=rKYTo6RtHxw&feature=youtu.be

jsweeney91, 2016-04-06:
@Boappp does it happen more often after 
1. hard punch 
2. begin roll
3. drop throttle

Boappp, 2016-04-06:
- Randomly throughout the riding styles (look my dropbox file with cleanflight blackbox).

- I already had the same problem on my old config (Just fc that does not change)

- for now I think it's a problem of the quality of fc, but i order a new spracing 20€ and naze32rev5 10€ banggood for a test...

Highoctane32, 2016-04-06:
Boappp your issue appears to be a loss of power, perhaps a bad connection? In your video right after the first issue happens you can hear at least one ESC reboot. So I would say it is likely one or more ESCs are loosing power

le3dmax, 2016-04-06:
I think it's ESC problems. I have 2 quads with exact same setups except
prop sizes. with 5" props flies great, no problem at all, the 4" props one
do this crazy flips for no reason. I went trough everything, but the only
thing I can think of is a fault ESC. One thing I noticed is that the
problem gets worse with high rates on Flip and roll.

jsweeney91, 2016-04-06:
@le3dmax it's not an esc issue this issue only happens on roll flips. Every other time it flies great

le3dmax, 2016-04-06:
Yes, in fact with me too. Roll flips and sometimes very aggressive yaw.

borisbstyle, 2016-04-06:
well thats exactly when a big change from low to high throttle is sent to
escs and those experience the most stress.

That only indicates that its an esc issue

Highoctane32, 2016-04-06:
Raise min throttle

jsweeney91, 2016-04-06:
but I don't experience any issues if i let the quadcopter drop and apply full throttle from min throttle

borisbstyle, 2016-04-06:
thats different....there power isnt applied as abrupt as on hard yaws

is your esc grounded?

jsweeney91, 2016-04-06:
yeah esc is grounded. There's no yaw input

Highoctane32, 2016-04-06:
You raising the throttle from min throttle to max in a free fall is different in a couple ways..one in theory you aren't able to command a throttle change as fast as the firmware does..2..air is flowing through the props auto rotating them..they want to spin in that scenario..it isn't until the descent is arrested that the load on the motors becomes significant by that point they are well above min throttle. This is a completely different scenario compared to trying to stop the rotation of a fast roll or flip.

Highoctane32, 2016-04-06:
inknow it seems counter intuitive but if a min throttle free fall were going to stall a motor it would do so during the fall, the effect of the airflow through the spinning props when falling isn't as taxing as some think IMO. also if it were we would would see higher amp draws during these maneuvers than we do on static bench testing, and I know people think that is the case but show me proof. I've never seen a higher amp draw from stopping a free fall with max throttle than I see on the bench with a restrained craft.

jsweeney91, 2016-04-06:
ah okay that makes sense, thank you for clarifying. The D term slows the
rate at which the FC can respond to changes, does that mean I should try
raising it?

Boappp, 2016-04-10:
I have resolved my problem, a pin (FC) on black wire of motor 2 was unsolder

closed 2016-06-13 by borisbstyle

enc0de101, 2017-02-19:
I have the same problem on all my quads. When i go for landing and drop the throttle it flips and goes crazy. 6 Quads please don't tell me faulty gear.

mattschinkel, 2017-10-26:
Mine was working good with PPM, when I changed to SBUS or IBUS it started flipping out randomly, usually during landing. I went back to PPM and it's good again, but I can only use 8 of my 10 channels.

When it did flip out, I wasn't able to put it in switch failsafe or disarm. After a few seconds, it would stop and I had to unplug & plug the battery to get it going again.

Does anyone else have this issue with this setup?
- flysky FS-I6x controller. 
- sp racing f3