betaflight/betaflight #298: Colibri Race flips and crashes

mavdnos, 2016-03-25:
2.5.4 brought a problem to the colibri race. after a blackflip it goes crazy and flipflops every way when its supposed to stop the flip. 2.5.3 is fine.

borisbstyle, 2016-03-25:
there is actually not much changed between 2.5.3 and 2.5.4 what could
affect this target. Are you sure it is not something else?

mavdnos, 2016-03-26:
It looks like its linked to acro pluss. allmost like its capped at a flip per second. if i go under that x flip per second it seems to be fine.

iconicj, 2016-03-26:
Did you try it on 2.5.3? There's a similar issue I'm having since 2.5.3  - If you came from a version before that, you might not have noticed it. This sounds exactly like what I'm experiencing.

SirVodkat, 2016-03-27:
I seem to be getting something similar (BF 2.5.3). Is this what is happening to you?
http://www.rcgroups.com/forums/showpost.php?p=34367489&postcount=23478

borisbstyle, 2016-03-27:
sounds more like an esc problem

Jbooost, 2016-04-18:
Having this same issue. Tried raising min command and still does it. On 2.6.1 https://www.youtube.com/watch?v=b2b9_vo19iw

rallyeverything, 2016-05-09:
Exact same issue on 2.6.1 

Quad tune is really solid/crisp well tuned in my opinion... but if i do back-to-back flips will eventually freak out and tumble.

- So far tried re-calibrating ESCs ( many times )
- Replaced ESC that had a different Min Throttle then others after calibration.
- Upped Min Throttle in Config.
- Lowered/Raised P terms
- Lowered looptime from 2k to 1k
- Took apart quad and verified all joints
- Oddly enough, found that my spek sat + wire was broken at the board and was sure i found the problem. re-soldered and wend for a triumphant test flight only to find the after a couple hard flips problem still there...

FWIW: it seems more likely to happen when I go to min throttle during the flip. Just flashed back to 2.5.4 to see if it helps, i;'ll try to get some BB data is problem persists.

RMRC DODO
Rotor Geeks 30A ESC BLH14.1

 Everything else feels so locked in I'm hoping to figure this out.

borisbstyle, 2016-05-09:
@Jbooost 
Are you using blheli 14.5?

mavdnos, 2016-05-10:
does it on 14.3   14,4 and 14.5 in my case

borisbstyle, 2016-05-10:
than your esc sound like faulty :)

That is the one and only reason for quad to flip as one of the motors
desyncs

Jbooost, 2016-05-10:
@borisbstyle  On whatever the powercube comes with in the Vendetta. I could update the escs but wasn't aware that there was a 14.5 target for the powercube escs.

drehmo, 2016-05-27:
I have a very similar issue on two Quads now.
I think (for me) it boils down to the LittleBees having a problem with Betaflight?
Just thinking, because I upgraded from my old DYS SN20s to 20A LittleBees yesterday and all out of a sudden, a very stable copter became a very unpredictable one.
It also randomly flips over.
Unfortunately I don't have the ability to Blackbox it.
What I did found though is a video of someone, with almost identical issues:

https://youtu.be/gcALtT8U0Eo (see 00:28 to 00:34 and 02:38 - 02:42)

simonsais, 2016-05-28:
I have kind of "after flip flips" - i make a flip very fast, and when i stop it immediately, it flips around a random axis (mostly diagonal) and hits the ground. When i smooth it out at the end of my flip, nothing happens. But it seems as the FC or ESC getting confuzed, and try to stabilize - but way later as needed.

phlakvest, 2016-06-22:
I think I'm having a similar problem with a Naze32 Rev5. I installed Little Bees running blHeli 14.5, and had a couple instances where in a fast double/triple roll the copter keeps rolling.

phlakvest, 2016-06-22:
I noticed the same issue on my F-303 with LittleBees (Blheli 14.5, Super_expo enabled, Airmode on, Rate 75). The first time I thought it was pilot error, but after the third time I looked for more info, and found this post.

I was able to capture it on video and blackbox if that helps. (The video/log has modified PIDs, but it happened on stock pids also.)

I think it actually was about to stabilize at the end, but I disarmed it too soon.

Blackbox: https://app.box.com/s/mbsqjs49zt9bhj09sed92eguab6dx3wl
Video: https://www.youtube.com/watch?v=S5CNZauLqq8

iconicj, 2016-06-22:
That's probably BLHELI 14.5, which had a really bad desync bug. Upgrading to 14.6 helped

Happy Connecting. Sent from my Sprint Samsung Galaxy S® 5

phlakvest, 2016-06-22:
Sorry, mistyped. I was running 14.6.

simonsais, 2016-06-22:
Never change a running system 😉

phlakvest, 2016-06-23:
Here is one without super_expo to rule that feature out. 

https://www.youtube.com/watch?v=t2bEW1aJJ90
https://app.box.com/s/72guy2pflbnkdjslmnknw901gkphwvh9

It's a new build, so unfortunately I can't rule out hardware, or step back before the problem started.

KiteAnton, 2016-06-23:
@phlakvest, by looking at your video it looks the same as a lot of these ESC "desync" problems. Your ESC/motor probably misses to keep the prop spinning enough and the small glitch is what you experience. Try raising your min_throttle and test by changing the anti_desync_power_step parameter (read more in the wiki).

phlakvest, 2016-06-24:
Good call kite. I had turned min throttle down so it would fall faster. I turned it back up to stock, and so far I haven't been able to duplicate the problems.

KiteAnton, 2016-06-27:
Problem seems to be fixed, closing.

closed 2016-06-27 by KiteAnton