betaflight/betaflight #301: Problem with gyro calibration on arm (2.5.4)

etracer65, 2016-03-26:
Before heading off to fly today I updated my fleet from 2.5.3 to 2.5.4 - mainly because I wanted the improved gyro calibration with faster loop times and the automatic calibration on arm.  All of my quads are x-frames with the battery on the bottom so I'm always careful about using stick command to recalibrate before arming and wanted the automatic option instead.

One of the quads was brand new and being maidened (Shen Drones Mixuko).  First couple of flights were fine.  On the third flight after arming and moving the throttle above min_check the quad flipped out and went into a high-speed yaw spin (tasmanian devil!).  Got it disarmed and it landed 25 feet away! After repairing the damage, tried again and it again flipped out but not nearly so violently.  Bench tested without props and the motors were all normal, sensors looked normal in the configurator, etc.  Triple-checked that the props were on correctly, and even changed them again to be sure there wasn't excessive vibration (they were new on the first flip-out).  Note that there were no crashes on this brand new quad before the first occurrence.  Since this was a new build, I begrudgingly assumed that I had some bad part or a problem with the flight controller (X-Racer F303 v2 with VBAT) and set it aside to fly my other quads.  Other specs: Littlebee 20a, Sunnysky 2204 2300kv, FrSky X4R-SB.  Completely default PID's (luxfloat), loop time 375 with accelerometer enabled.

First flight on the next quad was fine (RMRC Goby 180).  Next flight the quad did the same flip-out as soon as I got the throttle above min_check.  Intermittently it happened a couple more times.  Nearly identical config to first quad, just tuned PID's/rates.

So I moved on the the third quad (BeeRotor X200).  Same behavior, couple of good flights and then intermittent flip-outs on initial throttle up.  Once again, same config.

Having the same symptom on 3 different rigs negates a hardware problem possibility and the only common was that I upgraded all to 2.5.4.  So I went back to 2.5.3 and the problems completely went away.

Began to troubleshoot again later in the day and found that one of my flying buddies had also upgraded to 2.5.4 but wasn't having any problems (also with a X-Racer F303 FC).  The difference I found was that he was not using the auto calibration on arm.  So at the risk of more damage, I re-flashed back to 2.5.4 but this time set gyro_cal_on_first_arm = OFF.  No more problems.  Flew another 15 packs and everything was perfectly normal.

So I'm thinking there's a problem with the gyro calibration not completing completely or correctly when done during the arming sequence.  Thinking back on all the flip-outs each one was a little different in there intensity and although I can't prove it they all seemed to be in the yaw axis.

Possibly something unusual that may be contributing: I have motor_stop enabled and also have air mode enabled when I'm in rate mode and disabled in angle/horizon.  That way I can use angle mode for landings with lots of camera tilt and have motor stop working for easily killing the motors on landing (love this feature!).  Maybe there's some interaction with motor_stop and the new automatic gyro calibration on arm?

I normally take off in rate mode (with air mode enabled) so when I switch arm the motors spin.  The flip-outs occurred immediately after the throttle arched min_check (no other stick inputs).  I also tested in LOS in both angle and horizon mode and saw the same behavior.  Since I had motor_stop enabled the motors weren't spinning after armed and once throttle advanced past min_check the motors started spinning and the flip-out occurred.

etracer65, 2016-03-26:
Here's the config I was using.  I always flash new versions with a full chip erase and apply my delta below.  That way I only affect the settings I'm using.  Once I changed to gyro_cal_on_first_arm = OFF the problem went away (on 3 different quads).

serial 2 64 115200 57600 0 115200

feature RX_SERIAL
feature MOTOR_STOP
feature FAILSAFE
feature ONESHOT125
feature BLACKBOX
feature VBAT

set serialrx_provider = SBUS

set failsafe_delay = 10
set failsafe_off_delay = 10
set failsafe_throttle = 1150
rxfail 3 h
rxfail 4 s 1000

set min_throttle = 1065
set max_throttle = 2000
set min_command = 1000

set blackbox_device = SPIFLASH
set blackbox_rate_denom = 2

set gyro_lpf = OFF
set gyro_sync_denom = 3
set baro_hardware = NONE
set mag_hardware = NONE

set max_aux_channels = 4

set dterm_lpf_hz = 50

set yaw_deadband = 5

set gyro_cal_on_first_arm = ON

aux 0 0 1 1775 1825
aux 1 1 0 900 1200
aux 2 2 0 1300 1700
aux 3 27 0 1775 2100
aux 4 12 3 1375 1575
aux 5 25 3 1800 2100

adjrange 0 0 2 900 2100 12 2

set pid_controller = LUX

rateprofile 2

set rc_rate = 100
set rc_expo = 70
set rc_yaw_expo = 35
set thr_mid = 50
set thr_expo = 0
set roll_rate = 70
set pitch_rate = 60
set yaw_rate = 50

rateprofile 1

set rc_rate = 100
set rc_expo = 70
set rc_yaw_expo = 35
set thr_mid = 50
set thr_expo = 0
set roll_rate = 60
set pitch_rate = 50
set yaw_rate = 50

rateprofile 0

set rc_rate = 100
set rc_expo = 70
set rc_yaw_expo = 35
set thr_mid = 50
set thr_expo = 0
set roll_rate = 50
set pitch_rate = 40
set yaw_rate = 50

borisbstyle, 2016-03-31:
I will close this as it is solved in 2.6. Arming is prevented when gyro still calibrating

closed 2016-03-31 by borisbstyle

etracer65, 2016-03-31:
Awesome!  Thanks for this.  Now I just have to convince my flying buddies that it wasn't me and I really do know how to take off without immediately crashing! :)