betaflight/betaflight #303: failsafe kill switch throttle value

vytautas-valaitis, 2016-03-27:
If failsafe is enabled with kill switch (stage 2 enabled, kill switch enabled, land enabled) throttle value is set to "minimum length", but not to "throttle value used while landing".

etracer65, 2016-03-27:
This sounds normal.  From the help in the configurator for failsafe kill switch:

"Set this option to make the failsafe switch, configured in the modes tab, act as a direct kill switch, bypassing the selected failsafe procedure."

It's a kill switch, not an "activate failsafe landing" switch.

KiteAnton, 2016-06-23:
Old issue, closing since no bug.

closed 2016-06-23 by KiteAnton

amivit, 2017-02-08:
Wouldnt this be the same thing as disarming. What's the difference?

mikeller, 2017-02-08:
I think the 'kill switch' simulates an actual failsafe event, i.e. the craft cannot be armed again after it is triggered without being power cycled.

amivit, 2017-02-09:
Roger. I think you're right, unless the failsafe is configured to land by itself.