betaflight/betaflight #304: AR7700 and serial protocol on SPRacing F3

fporte, 2016-03-27:
Hi, I just replaced my Lemon RX ppm receiver by a AR7700 on a SP Racing F3 flashed with latest betaflight 2.5.4. I have a weird behavior when using serial protocol : erratic and brief uncontrolled pulses on random channels, mainly throttle. This can be clearly seen on the receiver tab as well as in flight. Any help would be greatly appreciated. Note : I don't have the issue in PPM mode.

Highoctane32, 2016-03-27:
Similar issues will happen if the wrong protocol for the rx is selected, i.e. If the rx is using DSMX set spek2048 in Cleanflight config. I

fporte, 2016-03-27:
Thx for the quick answer, I am using Spektrum2048 as parameter for the rx type and 11ms frame rate in my DX9.

fporte, 2016-03-27:
BTW I am using UART2 set up with serial mode.

Highoctane32, 2016-03-27:
From Spektrum website..
For sprf3 Connect the 3 pin servo connector on the 8 pin JST-SH cable to the IO2 port on the SP Racing F3 to the Remote Rx port on the AR7700 to run DS

fporte, 2016-03-27:
Yep, read that but I have no signal using this method, only port I found on and working was UART2.

seuffert, 2016-03-27:
as far as i understand you could use the SRXL port?!
https://github.com/cleanflight/cleanflight/issues/681

fporte, 2016-03-28:
That's right

fporte, 2016-03-28:
I give up for now, tried Spektrum1024, 22ms frame rate, UART3, still those damn random pulses... Returning to PPM :-(

seuffert, 2016-03-28:
could be an hardware issue? i will receive an AR7700 within the next days and i will test it on a spracingf3 - hopefully i can help

fporte, 2016-03-28:
Interesting! Let me know if you manage to have your AR7700 set up in SRXL mode talk to your F3 without issues. Could be indeed a hardware problem.

Cuxx, 2016-03-29:
My AR7700 & SPF3 works from the beginning > RemoteRX

fporte, 2016-03-29:
What port are you using?

fporte, 2016-03-31:
Must have been very tired the other day, I tried again with IO2 and RemoteRX and it worked like a charm. Thx all for your help.

closed 2016-03-31 by fporte