betaflight/betaflight #305: Broken wiki page: Betaflight-specific-CLI-commands

cmavr8, 2016-03-28:
While trying to revert an older version of a wiki page, it seems that I deleted it.
I will fix it asap.

For reference:
https://github.com/borisbstyle/betaflight/wiki/Betaflight-specific-CLI-commands/_history

cmavr8, 2016-03-28:
I can't push to the wiki:
`remote: Permission to borisbstyle/betaflight.wiki.git denied to cmavr8.
fatal: unable to access 'https://github.com/borisbstyle/betaflight.wiki.git/': The requested URL returned error: 403
`

Weird, since I could me direct changes from the web. Can someone please fix it?
The following two commits need to be reverted:
3be97484409e544662251514dd95137eb4120b33
d54c3d7b75c174994137223e2d59e3122684335f

Sorry for the mess... I was hoping github would let me fix it myself but it didn't happen :(

cmavr8, 2016-03-28:
I managed to fix it through github's GUI. Cumbersome but worked.

Sorry for the downtime :(

closed 2016-03-28 by cmavr8

waltr1, 2016-03-28:
Thanks.
Great you got it fixed.
Stuff happens but all good now.