betaflight/betaflight #307: Inverted rate profile selection

iconicj, 2016-03-29:
Setting up the rate profile selection via aux1, seems to be reversed in actual flight now.
I've got 1000-1500 for profile 1, 1600-2000  for profile 2. I've confirmed the location of the switch by the moving marker tab in the adjustments configuration tab. Down on the switch is slot1, up is slot 2.

The only difference between these profiles are pitch, roll, and yaw rates. Profile 1 uses rates of 0.2 - profile 2 uses rates of 0.5.

I've noticed in flight, switch down is the faster rates, which is backwards. 

Keep in mind I've already verified the marker location in the configuration tab. As a workaround, I simply invert the radio channel. So when switch is down (~1700) the slow profile is active, even though it's cut off is 1500. 

I noticed this behavior beginning in the first version which backed off to two profiles, but it was an easy work around.

iconicj, 2016-03-30:
[image]
[image]
[image]

iconicj, 2016-03-30:
I've attached photos of the setup I'm using.  With the switch down (not shown here, radio is off) but correctly showing on the left of the first bar, it should trigger profile 1.  

With the switch in mid or up position (its a 3 pos switch) it should trigger slot2.

In reality, this behavior is reversed, but it does not appear to be using slot2 at all.  With the switch down (attempting to use slot1) the rates of the quad are very slow - almost as if they are 0.  With the switch up, it's still slow, but feels more like it's using slot1.

I am wondering, when the 3rd profile was removed in betaflight, they may have been numbered in the code differently...  example profile0,1,2 whereas the configurator addresses them as 1,2, and 3.

borisbstyle, 2016-03-30:
@KiteAnton
Can you have a look?

KiteAnton, 2016-04-01:
This works differently then you expecting. Slot1 and slot2 has nothing to do with which profile is selected. It is the actual value of the last parameter "via channel" (in your case aux1) that selects which rateprofile that is selected. 

I just tried this myself and low values on aux1 sets rateprofile 0, medium values 1 and high values sets rateprofile 2.

I.e. there is nothing wrong with the firmware for this.

iconicj, 2016-04-04:
Since its always worked before, quite literally for two years, I just made the assumption something had changed.

Can you make a recommendation of the correct way a mode switch should be implemented?

KiteAnton, 2016-04-04:
Just setup a range for the same aux channel as you use for the via channel. Then it will change between the 3 rateprofiles depending on the actual channel value.

closed 2016-05-08 by iconicj