betaflight/betaflight #311: Channel issues with Naze32 rev6 & Radiolink AT9

stasky15, 2016-03-30:
Hi guys.

Today I received my first Naze32 rev6 board, but I have a HUGE issue: I can't set a correct Channel Map because just Aux channels works (1 to 4). No Roll, no Pitch, no Yaw and no Throttle.
I'm using PWM and the last firmware from Cleanflight (v1.12.0) for Naze32. But... if I use Cleanflight v1.9.0 or any Baseflight firmware: Roll, Pitch and Throttle works, but Yaw and all Aux channels (1 to 4) doesn't. If I use Betaflight the result it's like Cleanflight. No 4 main channels, just Aux channels works.
So, I suppose that's a software o setup problem, not hardware. I'm right?
SBUS didn't work either. With my old cc3d it does.

What's happening here? Thanks!

Here the's a pic: [https://goo.gl/photos/9aokPPu9kLxijWav6](url)

+here: [https://www.reddit.com/r/Multicopter/comments/4cjui0/issues_with_naze32_rev6_radiolink_at9/](url)

voroshkov, 2016-03-30:
I had exactly the same problem with Naze32 rev5, and it appeared to be the hardware issue.

Cleanflight versions higher than 1.9.0 seems to block first 4 channels if one of them doesn't work for some reason. Guess you should first check if RC signal reaches the Naze32 board, and then inspect if it flows to corresponding controller pins on the board. 

See a bit more details in my blog: [http://voroshkov.blogspot.com](http://voroshkov.blogspot.com.by/2016/02/most-of-last-weekend-i-spent-re.html)

borisbstyle, 2016-03-30:
increase max_aux channels perhaps
sec max_aux_channels = 8 or more if you have sbus

use mapping in cli
map 1234AETR for example

closed 2016-03-30 by borisbstyle