betaflight/betaflight #312: Motolab Tornado GPS

darrenm36, 2016-03-31:
Is it possible to have the GPS functionality added back for the Motolab target?
I notice most other targets have GPS enabled, but it's not defined in the motolab target.h file, is there a reason for not having it on the Motolab Tornado?

I use it for recording flight paths, altitude and speeds.  Good for both the OSD and for post flight analysis.

borisbstyle, 2016-04-05:
I am not sure if this will be added to the target. There is a inav branch what covers all the GPS features.

darrenm36, 2016-04-05:
Is there any reason it can't.  I've gone and compiled my own .hex file of 2.6.0 with gps enabled and it works, so the board seems to have the capacity and still runs at 4khz.  It'd be nice not to have to do this with each new build

borisbstyle, 2016-04-05:
@darrenm36 
It's not about the space. Moto Moto himself explicitely did the full target implementation. I would first need to know why he didnt choose to include GPS in it. I dont want to introduce unstability

darrenm36, 2016-04-05:
Fair enough, I'll just keep doing my own compiles then with GPS included.
Is there any thought of combining some of the inav features?  I read the developer and you talked about it at the beginning of the branch.  I love your code and how stable and agile my quad is, but also like some of the gps features, mapping, speed on OSD and especially RTH (even an inaccurate version) in case I lose signal.

P.s. great work on betaflight and all the new developments.  Loving it.

digitalentity, 2016-04-05:
Unfortunately top racing performance and GPS features gon't go together - position estimation in INAV is very taxing on CPU. However you are free to try INAV (I suggest trying out "testing" branch) and see it will fly good enough for you.

closed 2016-06-23 by KiteAnton