betaflight/betaflight #315: LUX yawing on flips

jsweeney91, 2016-04-01:
Happened a few times now, the LUX is yawing sometimes when I perform roll flips. You can see it at the end of this video:

https://youtu.be/l9WnHuEOgWU 

Failsafe is set to hold and I wasn't applying yaw

borisbstyle, 2016-04-01:
add more p to yaw. Might be too low perhaps

borisbstyle, 2016-04-01:
oh...looks like you dont have airmode enabled and you went 0 throttle

jsweeney91, 2016-04-01:
airmode was enabled for the entirety of the flight

borisbstyle, 2016-04-01:
def looks like not in there....your throttle stick seems to go below min
check where stabilisation stops.

Anyway on 2.6 low throttle stabilisation has been added back without
airmode

jsweeney91, 2016-04-01:
ah okay, I generally always drop throttle right down on flips. So it's strange that it only happened on that one

borisbstyle, 2016-04-01:
maybe that time airmode was disabled.

You have it on switch?

jsweeney91, 2016-04-01:
It's on a switch but because of the range I've set, I can't disable it

micaheli, 2016-04-02:
I'll go ahead and back you up with your yaw thing.... I just went back and
forth from lux to rewrite and it's only on lux.  Same exact quad config...
airmode on... I've both experienced it LOS and FPV.  I tried to get past it
with some yaw_deadband and yaw_expo, but other than making it super
slopping mid-stick, it didn't correct the odd yaw.

micaheli, 2016-04-02:
Knowing very little about this stuff, but going with my gut, I'd say
removing I-term on yaw may have caused this....

borisbstyle, 2016-04-04:
So whats about this issue? Sounds like tuning?

jsweeney91, 2016-04-05:
Updated to 2.6 will report back later today. I doubt its tuning

jsweeney91, 2016-04-05:
@micaheli btw when I said LUX I meant the flight controller not the PID

micaheli, 2016-04-05:
Oh, dang.. Thanks for clearing it up... Coincidentally, I am also running
the Lumenier LUX.

jsweeney91, 2016-04-05:
still happening on 2.6 borris

Highoctane32, 2016-04-05:
Do you run a yaw deadband?

jsweeney91, 2016-04-05:
yeah, yaw_deaband 6

jsweeney91, 2016-04-06:
https://www.youtube.com/watch?v=rKYTo6RtHxw&feature=youtu.be
 
3:45 

4:38

More unexpected behaviour on rolls

jsweeney91, 2016-04-06:
[dump.txt](https://github.com/borisbstyle/betaflight/files/205607/dump.txt)

dump of settings

jsweeney91, 2016-04-11:
replaced motors, ESC's and issue no longer present

closed 2016-04-11 by jsweeney91