betaflight/betaflight #320: yaw_iterm_reset_degrees default too low for tricopters

mikeller, 2016-04-03:
On 2.6.0, with `yaw_iterm_reset_degrees` at the default of 50, the tail servo on my tricopter was constantly oscillating when yawing left (yawing right was fine), with no amount of PID tuning getting rid of it. After I discovered `yaw_iterm_reset_degrees`, I tried setting it to 100, which immediately got rid of the oscillations when yawing.

Would it make sense to set a higher default for `yaw_iterm_reset_degrees`, or to do so when `mixer TRI` is set?

borisbstyle, 2016-04-03:
Yes good point.

The defaults are all optimized for quadcopters. Tricopters are different in
terms of yaw.

I will consider changing default for tricopters.
Any chance to get a blackbox log from this tricopter?

mikeller, 2016-04-03:
Does the tuning matter?
It is all over the show at the moment, after my attempts to tune, before I realised it was `yaw_iterm_reset_degrees` that was the problem.
It is dark right now, and the forecast is for bad weather for tomorrow, so it might be a couple of days.

borisbstyle, 2016-04-03:
well tricopter yaw has much more snappy behaviour and 50degrees threshold
is too low probably. The quadcopters have much worse yaw behaviour and the
iterm makes things worse when rate speeds get higher

mikeller, 2016-04-03:
Ah, my question was about the blackbox log, I wanted to know if a log from a tricopter that is not really PID tuned was any help to you.
But it's stormy outside now, so no chance to get a log today. Will take my tricopter to work tomorrow, and hope I'll get some time to do some indoor tuning / logging.

borisbstyle, 2016-04-03:
yes....doesnt matter. I only want to see that yaw behaviour you had with
reset at 50

mikeller, 2016-04-03:
Got the blackbox log for you:

https://drive.google.com/file/d/0B_6c1X6V1qkAaU5veFNyUkdHdUE/view

I also did some testing, and with everything up to `yaw_iterm_reset_degrees = 200` the servo did twitch at least once on very hard yaw left. Maybe `yaw_iterm_reset_degrees` should just be disabled for tricopters?

powdermnky007, 2016-04-21:
No problems on my tricopter.  In cleanflight/betaflight where all three motors are turning the same direction is dumb.  It works fine in latest betaflight with front two turning opposite directions, i have verified.  Below is an excerpt from Triflight setup guide.

The tricopter will fly better with the propellers mounted the following (Copter seen from above): left – CCW, right – CW, tail – CCW. This will help in coordinated turns. It’s not a big deal if front motors are swapped, with left rotating CW and right CCW. It will only mean that you will be fighting it a bit more with stick inputs in the turns. **However the CCW rotation of the tail motor is absolutely critical for proper triflight behavior. It will not fly well with the CW tail motor.**

Again I am using betaflight, and no special setup, just set my motors / props as they said to, and it flies great for me.  Give it a shot.  My tail is much better behaved than the old firmware version i was using (2.1.4)

mikeller, 2016-04-21:
Hi @powdermnky007.

What version of betaflight are you using, and what are your yaw PID settings?

Re propeller directions: I too mount the propellers with the front pair going in the opposite direction. For the tail prop, with betaflight (as opposed to triflight), that one can go either way, depending on rotation direction and how the servo is linked to the pivot, `smix reverse 5 2 r` will have to be used to reverse the direction the servo acts, or not.

My yaw PIDs are:

```
p_yaw = 60
i_yaw = 20
d_yaw = 20
```

Using these, I find that the default of `yaw_iterm_reset_degrees = 50` is too low, and the servo starts to oscillate when yawing very hard in one direction (but not the other, surprisingly). Setting

```
yaw_iterm_reset_degrees = 200
```

makes this a lot better. (It will still oscillate, but only for extreme yaw input.).

powdermnky007, 2016-04-21:
@mikeller flying the latest 2.6.1

`
set yaw_deadband = 5
set tri_unarmed_servo = OFF
set rc_smoothing = ON

set pid_controller = MWREWRITE
set p_pitch = 55
set i_pitch = 30
set d_pitch = 23
set p_roll = 55
set i_roll = 30
set d_roll = 23
set p_yaw = 70
set i_yaw = 40
set d_yaw = 0
set p_level = 40
set i_level = 40
set d_level = 50

set yaw_rate = 40
set tpa_rate = 20

set rc_rate = 100
set rc_yaw_expo = 40
`

have air mode turned on, no super expo

mikeller, 2016-04-21:
Interesting, your `i_yaw` is even higher than mine, and you don't get the oscillations. What is your `yaw_iterm_reset_degrees` set to?

powdermnky007, 2016-04-21:
whatever the default is.  I never messed with it.  its a 3s/4s 450ish tricopter - sunnysky 1400kv motors 8x45 crappy slowfly props.  Those settings work with 3s & 4s.  All I do is change the battery.  Use the same props too.

mikeller, 2016-04-21:
Would you mind looking up what it is? If yours is different to what I've got we know that this is the culprit.

powdermnky007, 2016-04-22:
set yaw_p_limit = 400
set pid_delta_method = MEASUREMENT
set dterm_lowpass_hz =  70.000
set dterm_average_count = 4
set iterm_reset_degrees = 200
set yaw_iterm_reset_degrees = 50
set yaw_lowpass_hz =  70.000

mikeller, 2016-04-23:
Done:

My PIDs with `yaw_iterm_reset_degrees = 50`:

https://www.youtube.com/watch?v=hOzevhYkITE

Your settings:

https://www.youtube.com/watch?v=6mm1s0u4Q4I

My PIDs with `yaw_iterm_reset_degrees = 200`:

https://www.youtube.com/watch?v=mY1QNH_eM2Y

powdermnky007, 2016-04-23:
Cool I'll get a video of mine tomorrow.  What exactly am I looking for in your videos?  I watched all three of them.  Sorry, they all look ok.  Don't see any constant oscillations. 

My tail servo acts like yours did on the first video.  When i stop the turn it moves to counteract it, but I think that is normal.  By increasing the yaw_iterm_reset_degrees to 200 that is saying once your copter is rotating 200 degrees per second on the yaw axis, it will reset the iterm.  So by setting it to 200, and just hovering you have effectively disable the iterm reset.  Your iterm is staying active at all times on yaw like in previous versions of the firmware.

mikeller, 2016-04-23:
Sorry the issue is not very well visible on the video. The oscillations only appear while the craft is yawing CW (but not when yawing CCW). More annoying than the oscillations themselves is that this causes the CW yaw to be much slower than the CCW yaw, which is not really something that can be shown in a video.
Your settings do not seem to be affected by this problem, but on my craft they have a very violent servo move when coming out of a yaw, for both directions of yaw. Setting `yaw_iterm_reset_degrees` to 200 does not seem to affect this.

powdermnky007, 2016-04-23:
I'm rendering video now & will upload it tonight.  My tricopter turns the same speed in both directions.  You should remove D from yaw.  Here is a release note from 2.6.0
_Removed D on yaw from all pid controllers. This saves the pid controller some calculations. Dterm on yaw isnt really needed_    Thats why I have my Yaw D set to zero.  I think its just ignoring your 20, I don't think thats the issue.  Just may as well set it to zero.  

I'm pretty sure the oscillation isn't anything to do with the I term.  Granted, I'm not a developer, and I can see from your contributions you know the code more than me.  I just fly.  But the I term isn't really known for oscillations and yours is really low, so I don't think an I of 20 would oscillate anyway (mine is 40).

this is your comment: _Your settings do not seem to be affected by this problem, but on my craft they have a very violent servo move when coming out of a yaw, for both directions of yaw. _
Interesting that my settings are not affected by the problem.  I believe the violent servo move when stopping a yaw is from my higher yaw P.  Yours is 60, mine is 70, makes sense.  Just for the fun on it, try lowering your yaw P to like 30 and see what that does.  That should make the servo react gentler at the end of the turn.

Around the 1:00 minute mark until the end of the video I start doing full speed yaw turns in either direction and you can see they are both about the same speed.  You can see at the end of each turn the servo goes full deflection in the opposite direction like you said.  Again I'm not an expert and may be wrong, but it seems like that is the correct thing for it to do.

When you first start a yaw, the servo goes full tilt the same direction.  Once your copter is up to the correct yaw speed, the servo goes back to center, as its just maintaining the current yaw speed.  When you center the stick, the servo has to move full tilt to the opposite direction to instantly stop the yaw move.  Once the yaw has stopped, the servo will center again.  You can see this behavior throughout the entirety of my video.  

Sorry if I'm taking this in the wrong direction, just trying to help. :)
My best guess for your yaw problem is too high P and too low I on yaw axis.  Try using my settings, as you said they worked fine and try these for the Yaw.  Best of luck.  
set p_yaw = 40
set i_yaw = 40

Also I use about 20 TPA as insurance that I don't have oscillations at full throttle, but TPA isn't active on the Yaw axis, just Pitch and Roll.  Just in case you didn't know ;)  So adjusting TPA won't help your yaw problems.  I'll post video tonight, let me know if you figure anything out.

powdermnky007, 2016-04-24:
should be live in an hour or so
https://www.youtube.com/watch?v=8wQeIECP79g&feature=youtu.be

mikeller, 2016-04-24:
Thanks @powdermnky007 , interesting video. The violent servo move after yawing is affecting your much larger and probably stiffer frame a lot less, on my frame it is strong enough to cause instability when coming out of turns.

I did another test flight with your settings, and this time I noticed the 'oscillations' when yawing CW with your settings as well, albeit much less than with my settings, probably due to the higher P value. (And by 'oscillations' I do not mean the servo oscillations that you commonly get on tricopters, it's more something that, whenever I yaw CW, the servo pivots the prop to effect the yaw, but then after .5 of a second or so, something resets the servo, and it goes back to mid position, only to pivot again, as the yaw control input is still present obviously.)

I had tried increasing `yaw_iterm_reset_degrees` to the maximum of 1000, but that made yaw very soft and uncontrolled. 200 to 300 seems to be the right range for my tricopter.

And you were right about D on yaw, changing that and setting it to 0 did not make a difference in behaviour at all. I'd had done a lot of tuning pre 2.6.0 in order to get rid of the servo oscillations that I'd had with my wobbly plastic frame, and a lot of that was down to the right D tuning, so I'd kept that when updating to 2.6.0.

As a last thought, what flight controller are you using? Clutching at straws, but this could be down to some sort of overrun on my F1 Naze32 rev6.

@borisbstyle : Do you want to add a different default for `yaw_iterm_reset_degrees` for tricopters, or leave it up to the owner to change this if they get this kind of oscillation on yaw?

powdermnky007, 2016-04-24:
I also use a naze 32.  Ok I have a guess at what is happening to you.  The I term keeps resetting. With a quad it doesn't matter as much, but with a tricopter it thrashes the servo around back and forth when you move in and out of that iterm reset area.  So that makes sense when raising iterm reset value would fix it.

Now that you mention it, I don't know if you can tell in my video, but when I'm yawing slowly I did notice my servo acting kinda weird, very small movements.  Nothing enough to make my quad move, but it was kinda spastic.

At the end of my video I started doing big fast yaws, and the servo acted perfectly on the full 360 fast yaws.

Maybe there needs to be a smoothing or a delay added in?  I don't know, Ill leave that up to Boris to recommend a fix.  Maybe he can check out triflight and see what they did.  I know tricopters are not his focus, but there might be a easy fix that can benefit quads as well.

powdermnky007, 2016-04-24:
Sorry, don't know how to edit post on phone. I Hebrew naze r5 and I see you have r6.  Try doing some vibration dampening on your fc mount.  The naze r6 and the lux fc (any fc with the same mpu as those use) can have noise and yaw twitches.

This might be amplifying your problem.  It might be easier just to change our your naze to an old r5 or just a new fc.

martinbudden, 2016-07-06:
@mikeller ,

can you state a suggested default value of `yaw_iterm_reset_degrees` for tricopters?

And for the record, can you state all your propeller directions?

mikeller, 2016-07-06:
The config that I am flying with 3.0.0 at the moment (and that is the most stable that I've ever seen for this tricopter):

```
set servo_pwm_rate = 400
set yaw_iterm_ignore_threshold = 1000
set p_yaw = 60
set i_yaw = 5
set d_yaw = 5
set yaw_rate = 60
```

The rest of the PIDs are default values from the firmware. (`yaw_iterm_ignore_threshold` seems to be quite crucial, for the default value or any low value i get a pretty bad 'waggle' of the tail whenever turning left.)

My motor directions are (from top):
- tail: CCW
- front right: CCW
- front left: CW
(i.e. front going 'inwards')

Thanks for looking into this.

borisbstyle, 2016-07-06:
I am in process redoing all the iterm stuff anyway. Requires a lot of
flight testing and time though.

mikeller, 2016-07-06:
Happy to help with that - tricopter or quad.

martinbudden, 2016-07-06:
@mikeller ,

have you tried setting up your front motors the other way, ie front left CCW, front right CW, this should give slightly better performance and way also reduce the waggle.

When you turn left, you bank left and yaw left. Banking left increases the speed of the front right motor and decreases the speed of the front left motor. But this motor speed change also produces a yaw effect - if a motor spins CW then speeding it up will produce a CCW yaw.

So if the front right motor spins CCW then speeding it up will produce a CW yaw, ie a yaw right. But for coordinated turns you want to yaw left when you bank left (ie front right motor speeds up). So the tail servo has to produce more yaw than otherwise.

This is why it is better to set it up front left CCW, front right CW.

borisbstyle, 2016-07-06:
I will release some test branches for that. I am also not 100% happu

mikeller, 2016-07-06:
@martinbudden, that's a very good point, and makes sense when thinking about it this way. I had tried swapping front motor directions at some stage when I first built the tricopter, but back then I had found that the spinning 'inwards' direction gave me slightly better results in terms of stability. But that was a while ago, and my tuning skills and probably also the quality of  firmware was a long way away from what they are now.

Will give this a go and report back when I get a chance to fly the craft during daylight hours.

closed 2016-07-09 by borisbstyle

mikeller, 2016-07-10:
Had a chance to take the tricopter for a spin during the day today. With the motor directions switched as per @martinbudden 's recommendation the 'wagging' of the tail servo is much reduced when yawing while hovering. Also, some testing showed that the amount of wagging is no longer reduced when setting `yaw_iterm_ignore_threshold` to a higher value.