betaflight/betaflight #322: Lux p.i.ds

dannygravity, 2016-04-04:
Just flashed my lux with 2.6.0 std lux float p.i.ds and on left in get twichting i.e jumping up and down as if it's getting power spickes so to speck.

borisbstyle, 2016-04-04:
Post blackbox logs.

dannygravity, 2016-04-04:
Sorry Boris only thing with Lux, it don't have black box il send a link to the vid in a second.

dannygravity, 2016-04-04:
http://youtu.be/Y8Cs43B4mVQ

dannygravity, 2016-04-04:
Link now up

borisbstyle, 2016-04-04:
A lot of folks seem to complain about lux lately. Can you try rewrite as well perhaps?

dannygravity, 2016-04-04:
Il give it a go thanks

Cheredanine, 2016-04-04:
Danny, had another guy complain about exactly the same on the lux FC thread on fpvlab, I got him to do a number of things, unfortunately he didn't do them one at a time but list is below and it fixed it for him (appreciate this doesn't diagnose the problem but it fixed it)

1. Set loop time as low as possible for your FC
2. Set yaw deadband
3. Recalibrate ESCs (in configurator, not blheli)
4. Recalibrate ESCs with oneshot disabled (in configurator, not blheli) and re enable oneshot after

Link below is to the thread, you can see on his vid he has the same issue with props off

http://fpvlab.com/forums/showthread.php?44971-The-Lumenier-LUX-Flight-Controller-STM32F303-32-bit-processor/page22

sirdoggg8u, 2016-04-05:
i also running beta 2.6 on naze rev6 on 2 of my quads ,with littlebe20 and getting severe twitching , tried all methods that i seen but cannot seem to get rid pls help

borisbstyle, 2016-04-05:
All on LUX and naze32 rev6 flight controllers here? It is really starting to show more and more that it all might me MPU6500 based

sirdoggg8u, 2016-04-05:
here the bb boris
[blackbox_log_2016-04-05_120319.TXT](https://github.com/borisbstyle/betaflight/files/204314/blackbox_log_2016-04-05_120319.TXT)

borisbstyle, 2016-04-05:
@sirdoggg8u 
It shows a lot of high frequency oscillation on roll and pitch and especially D is overreacting on that. That might also upset yaw axis perhaps.
Try removing D and see what it does. Just set D to 0 and make another log.

sirdoggg8u, 2016-04-05:
ok bud

borisbstyle, 2016-04-05:
@sirdoggg8u 
Also try "set gyro_lpf = 188hz" on this particular target or even lower values on these boards

sirdoggg8u, 2016-04-05:
here without gyro set to 188
[blackbox_log_2016-04-05_123704(d=0 r+p).TXT](https://github.com/borisbstyle/betaflight/files/204344/blackbox_log_2016-04-05_123704.d.0.r.p.TXT)

sirdoggg8u, 2016-04-05:
set to 188 and it spaz out uncontrollable ossilations

sirdoggg8u, 2016-04-05:
[blackbox_log_2016-04-05_125211(d=0 p+r 188.TXT](https://github.com/borisbstyle/betaflight/files/204356/blackbox_log_2016-04-05_125211.d.0.p.r.188.TXT)

borisbstyle, 2016-04-05:
@sirdoggg8u 
That gyro really seems completaly off with others what I have seen. There are occassional twitches in yaw and there is quite a lot of micro oscillation.

When removing D you also need to lower P usually. That explains those heavy oscillations with 188hz. Lower P and start raising those.

sirdoggg8u, 2016-04-05:
it isnt just this fc , getting with anuva rev6 , and so is my buddy , outta 3 quads all getting twitch 

shall i keep d at zero while lowering p and then upping again

borisbstyle, 2016-04-05:
Try a board with MPU6050! 

Naze32 rev6 uses exact same gyro as Lux FC. Both MPU6500.

Naze32 rev5 is also fine to compare. Put it on the same quad....enter same pids and see what it does

sirdoggg8u, 2016-04-05:
i have anuva quad running same setup as my qav210 and it running fine with naze32 rev6, no twitchs 
only difference is the motors and frame

borisbstyle, 2016-04-05:
@sirdoggg8u 
Some motors create more noise than others. I have several users who had these issues with certain motor/esc setups. They literally disable damping light and problems are gone on some setups. But also exchanging the board with MPU6050 gyro works too!

sirdoggg8u, 2016-04-05:
ok have to try that , if i can get old of naze rev5

borisbstyle, 2016-04-05:
But of course it would be nice to understand why these gyros/boards can show this behaviour or better how to solve it

sirdoggg8u, 2016-04-05:
i just disable dampen light and it solved issue

sirdoggg8u, 2016-04-05:
boris thx for ur time and ur help buddy ,

borisbstyle, 2016-04-05:
@sirdoggg8u 
Yup....that helped to 100 others, but still its something what needs more investigation as MPU6050 does not have problems with damped light

sirdoggg8u, 2016-04-05:
yep gonna find it bit different to fly that my others , never mind until its sort i will have no choice , thx once again boris

lampii, 2016-04-05:
Boris, we spoke on RCGroups (pawingo). I had a chance to play around yesterday. Hitting 2khz changed the behavior of the twitching, it was a bit softer but its still there. I setup my build with 2 flight controllers afterwards. The lux is wired in and driving the motors, a dodo is mounted on top solely to log and compare gyro output. Apart from these blackbox logs, is there anything else I can produce that will help you out?

borisbstyle, 2016-04-05:
That would be perfect!

dont forget to enable debugging for gyro

set debug_mode = GYRO

that will save raw gyro data to blackbox

lampii, 2016-04-06:
Would the tethered logging work? I think that might be easier than blackbox as long as I can arm while logging. 
Or is there a way to enable blackbox on power on without arming? I dont have a spare rx to connect atm.

lampii, 2016-04-06:
OK here are two logs, Both are lux logs one without DL and the other with.

It almost looks like motors get reversed!

I will work on getting the second board logging shortly.

[LUX NO DAMPED LIGHT.TXT](https://github.com/borisbstyle/betaflight/files/205619/LUX.NO.DAMPED.LIGHT.TXT)

[LUX-DAMPEDLIGHT.TXT](https://github.com/borisbstyle/betaflight/files/205608/LUX-DAMPEDLIGHT.TXT)

basdelfos, 2016-04-06:
I have the same issue on one of my quads. Higher looptime (125 vs 250) seems to help, the twitches occur less and seem less intensive. Strange thing is that only one quad has the problem. My QAV210 with LB20A has the twitching issue, but my EpiQuad 180X with DYS BL20A flies smooth without twitches. Both have LUX flightcontroller (bought at the same time) with same settings and DL enabled. 

I will do some blackbox logging also, maybe it helps out to compare logs from different quads to look for similarity in gyro traces.

borisbstyle, 2016-04-06:
@lampii 
This is your gyro. I would really say faulty gyro

[image]

lampii, 2016-04-06:
This would be a 2nd LUX to do this. I replaced mine the other day. 
Take a look at the 2nd log in the non-damped. Shows up fine. I didn't get much time yesterday, but I will generate some better longer ones.

dannygravity, 2016-04-06:
Is anyone flashing the lb esc throw the lux as I can't get mine to read to take off the damped option.

lampii, 2016-04-06:
Danny, check out the[ RCGroups lux thread](http://www.rcgroups.com/forums/showthread.php?t=2554204). All the passthrough info is there and links to VCP passthrough instructions/hex.

basdelfos, 2016-04-06:
This is my blackbox log from the LUX with damped light on.

[LOG00256.TXT](https://github.com/borisbstyle/betaflight/files/207452/LOG00256.TXT)

Also tried with damped light off and it was not flyable (or i did something wrong). Swapped the LUX from my EpiQuad over and doesn't seem to change anything, maybe even worse. Could this twitching be frame specific in combination with a LUX?
Last try for this day was to change back to the Naze rev6 that was previously in this frame, no issues at all! Go figure...

[blackbox_log_naze_rev6.TXT](https://github.com/borisbstyle/betaflight/files/207469/blackbox_log_naze_rev6.TXT)

If you compare the two logs, you will see that the gyro on yaw on the LUX has some frequency oscillations. I'm not sure if it is tuning related, but it seems that the gyro trace on yaw comes before the pid. Yaw gyro on the Naze rev6 seems to be very clean.

Both logs are just a hover indoor and with same PIDs and with DL on.

P.S. I did manage to fry my OpenLogger somehow, so have to wait until a new one arrives.

borisbstyle, 2016-04-06:
It's interresting to see the comparisation between these 2 boards with same
gyro and pids and to see that one of the 2 is not clean.

It's like something else is going on with those boards?

lampii, 2016-04-06:
Very interesting. Ok so i messed up on that log because I was holding the
quad and torquing the YAW with my hand to try and cause the jumps. Not sure
what the heck i was thinking... However i managed to pull a few more logs
just now, even with damped light disabled they looked the same with those
nasty spikes everywhere even though it hovers and sounds buttery smooth. I
then decided to reflash LUX and now the board seems to have gone nuts. It
mounts, then jumps to bootmode when plugged into usb so there goes that =(.
My naze6 logs look just as clean as yours.

lampii, 2016-04-06:
Instead of damped light, its in low. Not same capture, but same setup and gyro.. Is there a way to toggle blackbox on power on so i can skip wiring an RX? 

I guess it is a bad gyro. I can get a Dodo log as soon as I rip out the lux. Looks like 2 lux boards, likely same batch were bad.

Lux - 
[Lux-Low-bf2.6 .TXT](https://github.com/borisbstyle/betaflight/files/207515/Lux-Low-bf2.6.TXT)

NazeR6 -
[blackbox_log_naze2138.TXT](https://github.com/borisbstyle/betaflight/files/207518/blackbox_log_naze2138.TXT)

lampii, 2016-04-07:
I think the fact that it flies fine with a borked gyro and no damped light is a testament to your work. Thanks for looking into this. I will be sure to get the other guys to check their gyros this way.

basdelfos, 2016-04-07:
Would it help to use to fit the supplied rubber o-rings between the standoffs and the LUX?

basdelfos, 2016-04-07:
I solved my twitching problem with a little help from my Team MRO friends.
Placed the rubber o-rings between the standoffs and the LUX and lowered D on roll and pitch to 5. In the log you can see that D-term is much too active.

I hovered two batteries indoor without any visual twitching on yaw. It used to twitch so now and than every 5-10 seconds. Have to wait for the new logger to see the results in the gyro traces.

borisbstyle, 2016-04-08:
@basdelfos 
I see that more as a workaround. Of course D is the most sensitive part for noise. D of 5 is really low and is not consistent with other targets.

basdelfos, 2016-04-08:
I agree, but have to wait until my new openlogger comes in the mailbox to see what result these changes have in the logs. I will report back, hopefully in a week or so. I'm not sure if lowering the D to 5 did something to this solution, because I did not test with only the rubber o-rings.
Will do this test again with D back on 10 in the meantime my openlogger arrives.

What value of D on roll and pitch is more common, default is on 18 if i remember correct?

basdelfos, 2016-04-08:
Did a quick hover test with D back to 10 on roll and pitch. Still no visible twitching, so it looks like adding the o-rings solved the twitching for me. Will report back with a log as soon as possible.

basdelfos, 2016-04-17:
I had a few flights yesterday, it seems that the twitching is not totally gone. Still have twitches during hover, but not during flight anymore. With roll and pitch D on 10 I have more oscillations during flight. I club member has the same issues after flashing to 2.6.1 and a minor crash (for as long this is related). 

@borisbstyle, i read on the rcgroups threat that you are testing on a LUX board to get the gyro traces more clean. If you need some testing I'm happy to assist. Let me know.

BlackBear31, 2016-04-17:
Hi,
waste all my weekend on this issue...
QAV-R, Naze32 Rev6C, LittleBee 30A and XNOVA 2206-2500kv.
Saturday: BTF 2.6.1, Rewrite ou Luxfloat, ONESHOT125, looptime 1000 => no way to set correctly PID. Flotting or shacking. Impossible to set correctly. A nightmare...
Sunday: BTF2.6.1, Rewrite, esc flashed to Multishot, activate multishot on CLI, looptime on 250 => much better. I set the machine correctly (PID more or less what we can see on the net) but I still have some erratic twitching. I have disengaged "Damped Light" option on ESC and no more twitching... And just to finish the weekend my vidéo Tx has burnt...
I hope Boris that you will find a solution with these MPU6500...

Cheredanine, 2016-04-17:
@BlackBear31 
Boris has released a second alpha test release (see the rcgroups betaflight thread) which has some benefits for mpu6500 boards I believe.
Secondly if switching off damping light fixed the issue you could try turning it back on and changing the dither setting in blheli suite (try 7 or 31)

BlackBear31, 2016-04-17:
Ok I will test it. Thx. (I like eggs...)

grahamcropley, 2016-04-18:
I've been noticing this only when using the new EMAX RS2205 2300 and 2600kv motors. The only thing that seems to fix it is disabling damped light, or setting PWM output dither to highest.

Same results using known good FCs from my other quads Naze32 Rev 5 does the same as well as Rev 6, SPRacing F3, and F3 Mini, and Lux too. Tried Cleanflight, and Betaflight several versions. And BLHeli from 14.1 upwards.

Swapped out everything including regs, little bee 20A to DYS XM20A.

Tried P at 0.5 and both I and D to 0 to rule out PIDs. Doesn't fly, but could still reproduce moving it by hand.

Changing the motor timing had an effect but didn't cure it. Set to medium-low made the issue most obvious (to show people). Here is a video showing my symptoms. (Only right motors connected). I've got some blackbox logs I'll upload later. Difficult to tell if the motor is causing it and PIDs reacting, or gyro blipping causing the motor spaz.

http://youtu.be/LL9FwVnHYvQ

Finally gave up and put some of the original EMAX MT2204 2300kv motors with damped light and it flies smooth exact same setup and config and not a twitch in sight. Just underpowered.

Frame is a Tweaker FPV Addiction so 4inch props max. Would love got get the RS2205 2600kv working.

dannygravity, 2016-04-18:
I had the same problem with my lux on emax tried xnova no difference I than cut in half a heli canopy grommet and replaced it with the standard grommets fly 100% now  on STD lux pids on 2.6.1

sirdoggg8u, 2016-04-19:
I was the person that started this thread and turning off dampened. Light
solved the issue,  but i got so fed up cos i hated it being turned on with
my other quads and cant get used to it without it, so i turned it back on,
and would u believe i dont have a twitch at all,  and all i done is turn it
off flown 6 packs with it off, then turn back on,

Naze rev6
Emax 2205/2600
Littlebee 20a
Multishot

Scott

JasonB91, 2016-04-21:
A lot of "solutions" mentioned on this thread here. Not sure how many follow the Lux thread on RCG, but this was my fix, it might help some of you: 

I used these shock O-Rings top and bottom. 1 on bottom, or 2 on bottom was not good, the hard plastic nut on top (even loose) was transferring vibrations it seems. I ordered a BB for the Lux, but not here yet. 

http://www.amainhobbies.com/rc-cars-trucks/yokomo-x33-shock-x-ring-8-yokz2-s3x/p417710

You could also try these O-Rings, just need 2 packages of parts. Blue O-Rings inside the package. 

http://www.amainhobbies.com/rc-cars-trucks/traxxas-shock-rebuild-kit-tra2362/p45156
[image]

basdelfos, 2016-04-21:
@borisbstyle I've flown 4 packs with 2.6.2 alpha v3 today, with default PIDs and default dynamic_dterm_limit. No more twitching during hover, it flies so smooth now. Really like the dynamic dterm you implemented and most important, the twitching is also gone. I only noticed some prop wash, i read that lowering the dynamic_dterm_limit to 25 will mostly eliminate this. Will try that later.

SnipeXR, 2016-04-21:
Where's the link to 2.6.2 alpha like to try it.

basdelfos, 2016-04-21:
@SnipeXR, you can find it here: 

http://www.rcgroups.com/forums/showpost.php?p=34550089&postcount=25883

basdelfos, 2016-04-21:
And some more info: http://www.rcgroups.com/forums/showpost.php?p=34555761&postcount=25956

SnipeXR, 2016-04-21:
@basdelfos thanks saved me a ton of digging in rcgroups.

dunderburken, 2016-04-23:
@borisbstyle This is my solution on this problem. Setup: rs2205, LB30a, F3, MPU6500. Seem to have pretty darn heavy breaking. So powerful that it causes vibrations that the MPU6500 does not like. Symptoms: "out of sync motors" and heavy oscillations in flight. Solution 1) turn of active breaking (not an alternative), Solution 2) a tiny bit of vibration dampening tape when mounting FC - all running smooth again

[171 more replies]

closed 2016-07-09 by borisbstyle