betaflight/betaflight #324: dterm_lowpass_hz // CLI freezing

CreeT6, 2016-04-05:
set dterm_lowpass_hz use to be 0hz on default with 2.5.4 when i turn it on/ set it to 256hz or 100hz, quad flies better, but if i type dump in CLI, FC will freeze and needs a restart. however, quad still flies without freezing. same thing with 2.6, although the new default it 70hz.

borisbstyle, 2016-04-05:
it doesnt freeze here. I changed it all the time

CreeT6, 2016-04-06:
are you sure boris? this happens to 2 cc3d i tested, maybe this is a cc3d thing.  changing dterm filter and save then dump makes CLI freeze.

Arakon, 2016-04-06:
Wasn't there an issue with the yaw filter pointing to the dterm value in 2.6.0? 
PR #314 fixed it, but that's not in the current release hex, I believe.

borisbstyle, 2016-04-06:
It should work. I will double check on other targets, but honestly I dont
recommend to disable it at all.

You preferrably set it higher rather than lower or 0

borisbstyle, 2016-04-11:
Please try 2.6.1

closed 2016-04-11 by borisbstyle