betaflight/betaflight #325: Rate profiles: wrong matching?

sticksen, 2016-04-05:
Hey,

it seems that there´s a wrong matching of the rate profiles in the CLI and the GUI. Whenever I choose Profile 3 in the GUI, it switches back to profile 1. Also, if I´m making changes to profile 2 in the GUI, it seems to make changes to rateprofile 3 in the CLI.

Cheers

borisbstyle, 2016-04-05:
There is only profile 1 and profile 2 in betaflight

closed 2016-04-05 by borisbstyle

sticksen, 2016-04-05:
Ok, that´s alright. But why are there still 3 profiles in the CLI then? When I´m doing a 'dump all', there are 3 profiles in the log.

borisbstyle, 2016-04-05:
there are 3 rateprofiles and 2 profiles!

Thos 2 are different things!

sticksen, 2016-04-05:
Can you explain a bit more? I always thought that the rateprofile is linked to a profile. So Profile 0 uses rateprofile 0 etc.

borisbstyle, 2016-04-05:
all profiles again have 3 own rateprofiles