betaflight/betaflight #326: GPS don't work any times

dyter, 2016-04-05:
Sometime the GPS doesn't work and OSD coordinates disappear

http://www.hostingpics.net/viewer.php?id=273925IMG20160330162959.jpg
http://www.hostingpics.net/viewer.php?id=992631IMG20160330163113.jpg

borisbstyle, 2016-04-05:
@dyter 
Sorry but GPS is down on priority list in this project. CHeck inav branch from @digitalentity about better GPS support

dyter, 2016-04-06:
ok thank I tested it and it was a very good gps support. It's a all in one but without your gyro mod :(

aughey, 2016-05-05:
I'd like to try merging the GPS code into the main betaflight branch.  Any concerns with this?

dyter, 2016-05-06:
I run on Inavflight and it was very good for gps use + autopilot

KiteAnton, 2016-06-23:
Low priority for GPS support and no actual FW issue so I will close this issue.

closed 2016-06-23 by KiteAnton