betaflight/betaflight #327: AUX CHANNELS more than  4 doesn't work on SBUS

spacebubi, 2016-04-06:
Before on 2.4.1 release i used 5 AUX of my radio , on SBUS . Now i try to put the last 2.6.0 and when i put "set max_aux_channels = 5" on CLI and save the command was right but on TAB Receiver works only 4 AUX ch and not 5.
Some help about this issue?

lichtl, 2016-04-06:
Did you check on your transmitter is send more than 8 channels? Default on Taranis for example is to only send ch1 to ch8 so this needs to be adjusted as well.

MatsKarlsson, 2016-04-07:
Also there is a bug in the Taranis OpenTX 2.0.9, it's fixed in later versions.

spacebubi, 2016-04-08:
I resolved it...was the radio tx that after long session on sim. i've turned off...and returning to on it sets D1-8 and no D1-9 just before pulling off it. Yesss

closed 2016-04-08 by spacebubi