betaflight/betaflight #336: Not able to arm naze32 after a flush to betaflight

ggg22, 2016-04-10:
Working with Eachine Racer 250 (naze32 board).
When I use betaflight I am not able to arm it. It arms fine with the latest cleanflight.

Here is dump, status and tasks outputs.. 

Entering CLI Mode, type 'exit' to return, or 'help'

# dump

# version
# BetaFlight/NAZE 2.6.1 Apr  8 2016 / 22:44:36 (bdaf8ec)
# dump master

# mixer
mixer QUADX
mmix reset
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -ONESHOT125
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature RX_PPM
feature VBAT
feature FAILSAFE

# beeper
beeper GYRO_CALIBRATED
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
beeper BAT_LOW
beeper GPS_STATUS
beeper RX_SET
beeper ACC_CALIBRATION
beeper ACC_CALIBRATION_FAIL
beeper READY_BEEP
beeper MULTI_BEEPS
beeper DISARM_REPEAT
beeper ARMED
beeper SYSTEM_INIT
beeper ON_USB

# map
map AETR1234

# serial
serial 0 1 115200 57600 0 115200
serial 1 0 115200 57600 0 115200

# led
led 0 15,15:ES:IA:0
led 1 15,8:E:WF:0
led 2 15,7:E:WF:0
led 3 15,0:NE:IA:0
led 4 8,0:N:F:0
led 5 7,0:N:F:0
led 6 0,0:NW:IA:0
led 7 0,7:W:WF:0
led 8 0,8:W:WF:0
led 9 0,15:SW:IA:0
led 10 7,15:S:WF:0
led 11 8,15:S:WF:0
led 12 7,7:U:WF:0
led 13 8,7:U:WF:0
led 14 7,8:D:WF:0
led 15 8,8:D:WF:0
led 16 8,9::R:3
led 17 9,10::R:3
led 18 10,11::R:3
led 19 10,12::R:3
led 20 9,13::R:3
led 21 8,14::R:3
led 22 7,14::R:3
led 23 6,13::R:3
led 24 5,12::R:3
led 25 5,11::R:3
led 26 6,10::R:3
led 27 7,9::R:3
led 28 0,0:::0
led 29 0,0:::0
led 30 0,0:::0
led 31 0,0:::0

# color
color 0 0,0,0
color 1 0,255,255
color 2 0,0,255
color 3 30,0,255
color 4 60,0,255
color 5 90,0,255
color 6 120,0,255
color 7 150,0,255
color 8 180,0,255
color 9 210,0,255
color 10 240,0,255
color 11 270,0,255
color 12 300,0,255
color 13 330,0,255
color 14 0,0,0
color 15 0,0,0

# aux
aux 0 0 0 900 900
aux 1 0 0 900 900
aux 2 0 0 900 900
aux 3 0 0 900 900
aux 4 0 0 900 900
aux 5 0 0 900 900
aux 6 0 0 900 900
aux 7 0 0 900 900
aux 8 0 0 900 900
aux 9 0 0 900 900
aux 10 0 0 900 900
aux 11 0 0 900 900
aux 12 0 0 900 900
aux 13 0 0 900 900
aux 14 0 0 900 900
aux 15 0 0 900 900
aux 16 0 0 900 900
aux 17 0 0 900 900
aux 18 0 0 900 900
aux 19 0 0 900 900

# adjrange
adjrange 0 0 0 900 900 0 0
adjrange 1 0 0 900 900 0 0
adjrange 2 0 0 900 900 0 0
adjrange 3 0 0 900 900 0 0
adjrange 4 0 0 900 900 0 0
adjrange 5 0 0 900 900 0 0
adjrange 6 0 0 900 900 0 0
adjrange 7 0 0 900 900 0 0
adjrange 8 0 0 900 900 0 0
adjrange 9 0 0 900 900 0 0
adjrange 10 0 0 900 900 0 0
adjrange 11 0 0 900 900 0 0
adjrange 12 0 0 900 900 0 0
adjrange 13 0 0 900 900 0 0
adjrange 14 0 0 900 900 0 0

# rxrange
rxrange 0 1000 2000
rxrange 1 1000 2000
rxrange 2 1000 2000
rxrange 3 1000 2000

# servo
servo 0 1000 2000 1500 90 90 100 -1
servo 1 1000 2000 1500 90 90 100 -1
servo 2 1000 2000 1500 90 90 100 -1
servo 3 1000 2000 1500 90 90 100 -1
servo 4 1000 2000 1500 90 90 100 -1
servo 5 1000 2000 1500 90 90 100 -1
servo 6 1000 2000 1500 90 90 100 -1
servo 7 1000 2000 1500 90 90 100 -1

set mid_rc = 1500
set min_check = 1040
set max_check = 1900
set rssi_channel = 0
set rssi_scale = 30
set rssi_ppm_invert = OFF
set input_filtering_mode = OFF
set rc_smoothing = OFF
set roll_yaw_cam_mix_degrees = 0
set max_aux_channels = 6
set debug_mode = NONE
set min_throttle = 1150
set max_throttle = 1850
set min_command = 1000
set servo_center_pulse = 1500
set 3d_deadband_low = 1406
set 3d_deadband_high = 1514
set 3d_neutral = 1460
set 3d_deadband_throttle = 50
set use_oneshot42 = OFF
set use_multishot = OFF
set motor_pwm_rate = 400
set servo_pwm_rate = 50
set disarm_kill_switch = ON
set gyro_cal_on_first_arm = OFF
set auto_disarm_delay = 5
set small_angle = 25
set fixedwing_althold_dir = 1
set reboot_character = 82
set serialrx_provider = SPEK1024
set sbus_inversion = ON
set spektrum_sat_bind = 0
set spektrum_sat_bind_autoreset = 1
set telemetry_switch = OFF
set telemetry_inversion = OFF
set frsky_default_lattitude =  0.000
set frsky_default_longitude =  0.000
set frsky_coordinates_format = 0
set frsky_unit = IMPERIAL
set frsky_vfas_precision = 0
set frsky_vfas_cell_voltage = OFF
set hott_alarm_sound_interval = 5
set battery_capacity = 0
set vbat_scale = 110
set vbat_max_cell_voltage = 43
set vbat_min_cell_voltage = 33
set vbat_warning_cell_voltage = 35
set vbat_hysteresis = 1
set vbat_pid_compensation = OFF
set current_meter_scale = 400
set current_meter_offset = 0
set multiwii_current_meter_output = OFF
set current_meter_type = ADC
set align_gyro = DEFAULT
set align_acc = DEFAULT
set align_mag = DEFAULT
set align_board_roll = 0
set align_board_pitch = 0
set align_board_yaw = 0
set max_angle_inclination = 700
set gyro_lpf = OFF
set gyro_sync_denom = 8
set gyro_lowpass_hz =  80.000
set moron_threshold = 32
set imu_dcm_kp = 2500
set imu_dcm_ki = 0
set alt_hold_deadband = 40
set alt_hold_fast_change = ON
set deadband = 0
set yaw_deadband = 0
set throttle_correction_value = 0
set throttle_correction_angle = 800
set yaw_control_direction = 1
set yaw_motor_direction = 1
set yaw_jump_prevention_limit = 200
set tri_unarmed_servo = ON
set servo_lowpass_freq = 400
set servo_lowpass_enable = OFF
set super_expo_factor = 30
set failsafe_delay = 10
set failsafe_off_delay = 10
set failsafe_throttle = 1000
set failsafe_kill_switch = OFF
set failsafe_throttle_low_delay = 100
set failsafe_procedure = OFF
set rx_min_usec = 885
set rx_max_usec = 2115
set gimbal_mode = NORMAL
set acc_hardware = AUTO
set acc_lpf_hz =  10.000
set accxy_deadband = 40
set accz_deadband = 40
set acc_unarmedcal = ON
set acc_trim_pitch = 0
set acc_trim_roll = 0
set baro_tab_size = 21
set baro_noise_lpf =  0.600
set baro_cf_vel =  0.985
set baro_cf_alt =  0.965
set baro_hardware = AUTO
set mag_hardware = AUTO
set mag_declination = 0
set pid_process_denom = 1
set blackbox_rate_num = 0
set blackbox_rate_denom = 0
set blackbox_device = SERIAL
set magzero_x = 0
set magzero_y = 0
set magzero_z = 0

# rxfail
rxfail 0 a
rxfail 1 a
rxfail 2 a
rxfail 3 a
rxfail 4 h
rxfail 5 h
rxfail 6 h
rxfail 7 h
rxfail 8 h
rxfail 9 h
rxfail 10 h
rxfail 11 h
rxfail 12 h
rxfail 13 h
rxfail 14 h
rxfail 15 h
rxfail 16 h
rxfail 17 h

# profile
profile 0
############################# PROFILE VALUES ####################################
profile 0

set yaw_p_limit = 400
set pid_delta_method = MEASUREMENT
set dterm_lowpass_hz =  70.000
set dterm_average_count = 4
set iterm_reset_degrees = 200
set yaw_iterm_reset_degrees = 50
set yaw_lowpass_hz =  70.000
set pid_controller = MWREWRITE
set p_pitch = 45
set i_pitch = 30
set d_pitch = 18
set p_roll = 45
set i_roll = 30
set d_roll = 18
set p_yaw = 90
set i_yaw = 40
set d_yaw = 0
set p_alt = 50
set i_alt = 0
set d_alt = 0
set p_level = 50
set i_level = 50
set d_level = 100
set p_vel = 55
set i_vel = 55
set d_vel = 75
rateprofile 0

# rateprofile
rateprofile 0

set rc_rate = 100
set rc_expo = 60
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 50
set pitch_rate = 50
set yaw_rate = 50
set tpa_rate = 0
set tpa_breakpoint = 1500

# status
System Uptime: 85 seconds, Voltage: 0 * 0.1V (1S battery - NOT PRESENT), CPU:3%
CPU Clock=80MHz, GYRO=MPU6050, ACC=MPU6050.n, BARO=BMP280, MAG=HMC5883
Cycle Time: 1004, I2C Errors: 0, config size: 1300

# tasks
Task list:
0 - (SYSTEM) max: 3us, avg: 0us, rate: 10hz, total: 1ms
1 - (GYRO/PID) max: 484us, avg: 253us, rate: 992hz, total: 22sec
2 - (ATTITUDE) max: 302us, avg: 270us, rate: 99hz, total: 2sec
3 - (ACCEL) max: 373us, avg: 133us, rate: 332hz, total: 4sec
4 - (SERIAL) max: 639313us, avg: 2us, rate: 99hz, total: 745ms
5 - (BEEPER) max: 7us, avg: 0us, rate: 99hz, total: 13ms
6 - (BATTERY) max: 87us, avg: 1us, rate: 49hz, total: 25ms
7 - (RX) max: 77us, avg: 44us, rate: 49hz, total: 300ms
8 - (COMPASS) max: 127us, avg: 93us, rate: 9hz, total: 104ms
9 - (BARO) max: 123us, avg: 60us, rate: 7575hz, total: 606ms
11 - (ALTITUDE) max: 246us, avg: 140us, rate: 39hz, total: 517ms

#

ggg22, 2016-04-10:
also, unrelated but thought I mention it.. the documentation link in the gui seems broken "Documentation for 2.6.1" (upper right corner)
https://github.com/cleanflight/cleanflight/tree/v2.6.1/docs

lichtl, 2016-04-10:
Have you checked and adjusted rxrange? Channel 1-4 should go from 1000-2000. If not it is possible that your throttle value is just to high.

ggg22, 2016-04-10:
thanks for your suggestion, lichtl. 
Have just entered the following values via CLI and now I get values 1000-2000 for first 4 channels when moving the sticks to the end of their ranges:

# rxrange
rxrange 0 980 1986
rxrange 1 980 1986
rxrange 2 986 1992
rxrange 3 1084 1889

Interestingly I didn't have to enter any of these values when using the Cleanflight version and it was arming fine.
I am using mode 1 transmitter and used to arm when moving yaw to the max right with throttle to min.
I get one longer followed by 3 short green led blinks when I get the left stick in lower left corner. Not sure what it means but doesn't seem like arming since motors don't spin (and don't have MOTOR_STOP enabled)

thanks again!

borisbstyle, 2016-04-11:
Cleanflight has higher min_check threshold. Betaflight has less deadband 

cleanflight = 1100
betaflight = 1040

Please make sure that your transmitter reaches 1000-2000 as stated in any manual. If your receiver can't reach that raise the min check

closed 2016-04-11 by borisbstyle

ggg22, 2016-04-11:
Thanks borisbstyle.

Using Spektrum Dx7s. Played with the servo settings and it doesn't seem I can get the outputs in the range 1000-2000 without using the rxrange commands.
When I use the rxrange commands as follows I see the transmitter sticks move in the 1000-2000 range. Not sure if this is what is required.

rxrange 0 980 1986
rxrange 1 980 1986
rxrange 2 986 1992
rxrange 3 1084 1889

Also, I raised the min_check to 1100 but again no luck.

ggg22, 2016-04-14:
Have tried ARM mode and an aux channel to arm it but still no luck. This would suggest it's not the stick range values but something else.

ggg22, 2016-04-14:
Have just tried a different receiver.. still no luck.. would appreciate if anybody can help with this.

borisbstyle, 2016-04-15:
check your throws

ggg22, 2016-04-15:
Thanks borisbstyle.
With the new receiver (AR8000) I managed to get the switch ARM to work. 
That was not working with the previous receiver (OrangeRX).
Did not make any configuration adjustments.
That made me curious if the stick arming will work now so I removed the switch arming and YES IT WORKS!
Yesterday when I connected the new receiver I had the stick controls all as expected but didn't have the AUX channels all correct. Fixed that today and used Spectrum channel map in the receiver tab. No other changes were done and it seemed to have worked. No idea what was missing before.

Ry40, 2016-04-20:
i am having the same problem trying to arm my powercube, has no problem arming with clean flight but with betaflight it will not arm.

borisbstyle, 2016-04-20:
check your radio endpoints.

min_check is lower in betaflight.

This is all documented on wiki

Ry40, 2016-04-21:
ahhhhh, sorry and thank you,

djsomi, 2016-05-16:
Default 1080 was too low for my radio (jumped around 1090) so set min_check = 1100 is solved the arming issue for me also.
Big Thx!

Bobby78, 2017-01-22:
I'm having the same issue with beta flight and spektrum I can't get it to arm any reason why?

mikeller, 2017-01-22:
- compare `min_check` with throttle received from your TX (throttle must be below `min_check` to arm);
- check load (bottom of configurator). If close to / above 100 it will prevent arming.

Bobby78, 2017-01-22:
How do I do that?  I'm used to cleanflight with cleanflight I only have to adjust the trim on my radio .  I'm clueless

mikeller, 2017-01-22:
Read https://github.com/cleanflight/cleanflight/blob/master/docs/Controls.md

Bobby78, 2017-01-22:
So if I check on the reciever on betaflight my throttle min is 1017 and mid is 1402 and high is 1900.  Where do I go from there ?

Bobby78, 2017-01-22:
Mikeller.. I have read that a couple times already and am still clueless.  I belive I found the minimum it's at 1017

pkerney, 2017-06-14:
I went to my receiver tab and looked at the min/max values for throttle that were being reported by the display. I got 1109-1732. So I went to CLI and entered
set min_check = 1150
set max_check = 1700
save
Bingo. I seems to want you to move the throttle stick to exceed the min/max values and then once that has been done you can use the right rudder to arm.
PS. I'm a novice at this too.

mikeller, 2017-06-14:
@pkerney: Probably a better way of getting the same result would be to configure the output of your TX to go full scale from 1000 to 2000. This will give you better resolution.

pkerney, 2017-06-14:
Thanks @mikeller Agreed.
I was given a HobbyKing 4-channel toy TX+RX combo in this quad to get working for someone and it has no capabilities at all, not even AUX channels let alone end-point adjustments, sub trim, rates, expo etc.
Posted on this thread as others may be in the same position with low end non-configurable radio gear.
(Only a novice at Naze32/Cleanflight, experienced on other stuff. Don't want to install any of my good radio gear into it.)