betaflight/betaflight #338: Makefile isn't working when executed in terminal.

SiddhaKilaru, 2016-04-11:
Hello, whenever I try to run the Makefile I get this...

 %% startup_stm32f10x_md_gcc.S
make: arm-none-eabi-gcc: No such file or directory
make: *** [obj/main/NAZE/startup_stm32f10x_md_gcc.o] Error 1

Any ideas why?

nathantsoi, 2016-04-11:
you need a toolchain: https://github.com/cleanflight/cleanflight/blob/master/docs/development/Building%20in%20Ubuntu.md or https://github.com/cleanflight/cleanflight/blob/master/docs/development/Building%20in%20Mac%20OS%20X.md

SiddhaKilaru, 2016-04-11:
Hello @nathantsoi, thanks for the reply!  I chose the second link and it says,

"Now you just need to add the bin/ directory from inside the GCC directory to your system's path. Run nano ~/.profile. Add a new line at the end of the file which adds the path for the bin/ folder to your path, like so:

export PATH=$PATH:~/development/gcc-arm-none-eabi-4_9-2015q2/bin

Press CTRL+X to exit nano, and answer "y" when prompted to save your changes."

I'm kinda confused at this point. What are they exactly telling us to do?

nathantsoi, 2016-04-11:
if you're on a mac, you should just install the toolchain via [homebrew](brew.sh)

```
brew tap PX4/homebrew-px4
brew update
brew install gcc-arm-none-eabi
```

that will fix your path problem.

while you're at it, could you update [the wiki](https://github.com/borisbstyle/betaflight/wiki)?

SiddhaKilaru, 2016-04-11:
Hey @nathantsoi , entering that did fix my problem. Also, entering nano .bash_profile instead of nano ~/.profile worked as well. What part of the Wiki should I edit, or should I create another section?

nathantsoi, 2016-04-11:
new page maybe? like "Mac OS X Development Setup"

SiddhaKilaru, 2016-04-11:
Alright I'll start working on that @nathantsoi. Thanks for the help

closed 2016-04-11 by SiddhaKilaru