betaflight/betaflight #340: Airmode

Spudspitter, 2016-04-12:
this is an odd one. By they way i came from open pilot and betaflight is just down right the best thing i have ever flown. The problem is air mode. when I drop throttle to zero I have roll and pitch control but i have no yaw control. any ideas. oh and this does not happen all time. for example I went through 4 or 5 lipos yesterday it worked and it was magic. but then after charging up the lipos and going back out aprox 3 hours later I whent for a nice flight droped the throttle to spin back a float backwards no yaw control. she flys great with throttle up yaw is perfect. I am using a cc3d evo littlebee 20A flashed with the 14.4 BlHeli. and emax 2204 black bottomes reds are on there way. any ides. If you need my PID i can get them i just dont see how thats the problem. I feel its something in the min throttle or something. someone help me I love air mode when it works it is the best drug since fpv goggles. lol

KiteAnton, 2016-04-12:
Do you stick arm? If so you will have no yaw below min_check. 

Use switch arming instead, that will solve your problem.

Spudspitter, 2016-04-12:
Even after I start flight. I thought that when betaflight sees throttle and roll or pitch min_check is disabled.  And remember sometimes it qorks some times it doesn't.

borisbstyle, 2016-04-12:
To get yaw control on zero throttle you need to move to switch arming. Think about why.....otherwise you would never be able to disarm ;)

closed 2016-04-12 by borisbstyle

Spudspitter, 2016-04-12:
I am not worthy.  Thanks so much. Kind of obvious now that think about it. I moved over to switch arming and I have no problems.