betaflight/betaflight #343: Low Voltage Alarm

MBR595, 2016-04-13:
Hi,

I'm using BF 2.6.1 on a NAZE32 Rev5 board. Once I upgraded the FW my voltage alarm will start beeping around 15.5V. I swapped batteries and verified voltage readings matched on the OSD and portable voltage alarm. I'm not sure if this is a glitch in the FW or a HW issue. I never had this problem on BF 2.2 so I'm guessing it's a FW issue. Any ideas anyone?

Arakon, 2016-04-13:
I had a similar issue. When my MWOSD was set up to use the FC voltage alert levels, it showed low voltage at 11.6v (3S pack) or something already, despite the "low" cell value being 3.4v.

MBR595, 2016-04-13:
Arakon...were you able to fix it?

nathantsoi, 2016-04-13:
i had the same issue w/ high voltage lipos, try setting `set vbat_max_cell_voltage = 445` in the cli

MBR595, 2016-04-13:
I'll give that a try

MBR595, 2016-04-13:
Nathantsoi....were you using 4S?

nathantsoi, 2016-04-13:
4s high voltage lipos (charged to ~4.42v per cell). my issue was not caused by the number of cells, but the voltage of each cell was greater than the default of `4.3v`. if each cell is over `vbat_max_cell_voltage` the fc thinks there are more cells than there actually are and therefore the total voltage / total cells is "low"

MBR595, 2016-04-13:
Hmmm...I normally charge my lipos @ 4.2V/cell which would put them under the 4.3V default. I don't know if this would work but I'll try.

nathantsoi, 2016-04-13:
yeah, this wont make a difference then. for some reason i thought u said they were charged over 4.2v. did you measure the voltage w/ your flight controller and get 11.6v or with a different battery tester? maybe `vbat_scale` is wrong?

MBR595, 2016-04-13:
I measured with a different battery tester and the osd. The tester and osd match. My vbat_scale is default at 110.

Arakon, 2016-04-14:
I'm using normal LiPos and have this exact same setup running with no issue on older BF versions (2.4.1). My fix ended up being that I set the alarm voltage manually in the OSD rather than using the FC data.

MBR595, 2016-04-14:
Arakon....so you just turned off the voltage indicator in CF?

KiteAnton, 2016-06-23:
@MBR595 & @Arakon, is this issue still active ? Is the problem still present in latest 2.8.1 sw? Otherwise please close your issue.

Arakon, 2016-06-24:
It seems to be acting normal in 2.7.1 and 2.8.1.

KiteAnton, 2016-06-24:
Ok good. I close this, reopen or create new if problem occurs also in 2.8.1 or later.

closed 2016-06-24 by KiteAnton