betaflight/betaflight #344: MPU6000 SPI Clock Divider

kc10kevin, 2016-04-13:
I have a question concerning the MPU6000 SPI code.   I put a STM32F303 on a CC3D and made a target (also have a board I created with SPI MPU6000).   Everything work great, but when I go to 8k (125) the CPU maxes out at 100% even with the ACC disabled.  On a test board with the 6500 SPI it was not happening  It was driving me crazy why this was happening, so I went through the MPU6000 gyro code. 

SpiGyroInitInt calls mpu6000AccAndGyroInit.   it appears that the spi clock is slowed down for slow register writes, then speed back up 18mHz.   When it returns, to SpiGyroInitInt the clock is immediately slowed down again, but never speed back up.   I inserted the following at the end of SpiGyroInitInt and all is well.  CPU nice and low.   Not sure If this is a hack and not right.   Can I get some input?

spiSetDivisor(MPU6000_SPI_INSTANCE, SPI_18MHZ_CLOCK_DIVIDER);  // 18 MHz SPI clock

Code with added line (sorry about the formatting below.  I must be an idiot since I can't figure out the "Insert code" option):

void mpu6000SpiGyroInit(uint8_t lpf)
{
    mpuIntExtiInit();

    mpu6000AccAndGyroInit();

    spiSetDivisor(MPU6000_SPI_INSTANCE, SPI_0_5625MHZ_CLOCK_DIVIDER);

    // Accel and Gyro DLPF Setting
    mpu6000WriteRegister(MPU6000_CONFIG, lpf);
    delayMicroseconds(1);

    int16_t data[3];
    mpuGyroRead(data);

    if (((int8_t)data[1]) == -1 && ((int8_t)data[0]) == -1) {
        failureMode(FAILURE_GYRO_INIT_FAILED);
    }
    
    spiSetDivisor(MPU6000_SPI_INSTANCE, SPI_18MHZ_CLOCK_DIVIDER);  // 18 MHz SPI clock

}

static void mpu6000AccAndGyroInit(void) {

    if (mpuSpi6000InitDone) {
        return;
    }

    spiSetDivisor(MPU6000_SPI_INSTANCE, SPI_0_5625MHZ_CLOCK_DIVIDER);

    // Device Reset
    mpu6000WriteRegister(MPU_RA_PWR_MGMT_1, BIT_H_RESET);
    delay(150);

    mpu6000WriteRegister(MPU_RA_SIGNAL_PATH_RESET, BIT_GYRO | BIT_ACC | BIT_TEMP);
    delay(150);

    // Clock Source PPL with Z axis gyro reference
    mpu6000WriteRegister(MPU_RA_PWR_MGMT_1, MPU_CLK_SEL_PLLGYROZ);
    delayMicroseconds(15);

    // Disable Primary I2C Interface
    mpu6000WriteRegister(MPU_RA_USER_CTRL, BIT_I2C_IF_DIS);
    delayMicroseconds(15);

    mpu6000WriteRegister(MPU_RA_PWR_MGMT_2, 0x00);
    delayMicroseconds(15);

    // Accel Sample Rate 1kHz
    // Gyroscope Output Rate =  1kHz when the DLPF is enabled
    mpu6000WriteRegister(MPU_RA_SMPLRT_DIV, gyroMPU6xxxGetDividerDrops());
    delayMicroseconds(15);

    // Gyro +/- 1000 DPS Full Scale
    mpu6000WriteRegister(MPU_RA_GYRO_CONFIG, INV_FSR_2000DPS << 3);
    delayMicroseconds(15);

    // Accel +/- 8 G Full Scale
    mpu6000WriteRegister(MPU_RA_ACCEL_CONFIG, INV_FSR_16G << 3);
    delayMicroseconds(15);

    mpu6000WriteRegister(MPU_RA_INT_PIN_CFG, 0 << 7 | 0 << 6 | 0 << 5 | 1 << 4 | 0 << 3 | 0 << 2 | 0 << 1 | 0 << 0);  // INT_ANYRD_2CLEAR
    delayMicroseconds(15);

#ifdef USE_MPU_DATA_READY_SIGNAL
    mpu6000WriteRegister(MPU_RA_INT_ENABLE, MPU_RF_DATA_RDY_EN);
    delayMicroseconds(15);
#endif

    spiSetDivisor(MPU6000_SPI_INSTANCE, SPI_18MHZ_CLOCK_DIVIDER);  // 18 MHz SPI clock
    delayMicroseconds(1);

    mpuSpi6000InitDone = true;
}

lkaino, 2016-04-15:
I don't have the answer to your question but am really interested about this.

So do you think that the MPU6000 is always used at the slower speed after initialization in CF?

kc10kevin, 2016-04-15:
I don't have a debugger set up and am running the code line by line.  I'm not 100% sure, so that's why I am asking the question here and hopefully the betaflight experts can chime in.  

It does appear that the clock is slowed down, but never speed back up after initialization is complete.   If that is true, the issue may not have become apparent since the only target with an SPI MPU6000 was the CC3D up until recently.  No one had really tried to push the CC3D since it had an F1.

borisbstyle, 2016-04-15:
I have many different SPI boards over here and they run fine with 125us sampling. Even with acc enabled with minimal features it works.

borisbstyle, 2016-04-15:
These are the ones I tried and using same driver:
Cyclone
LUX
COLIBRI_RACE

kc10kevin, 2016-04-15:
If I'm not mistaken, the cyclone is the only one with the 6000.  The other two have the 6500 which runs perfectly on my board.  So the Cyclone has no problem at 125us?  If so, then I'm really stumped.   The CC3DF3 will not run at 125us.

My board with exact same layout, just MPU changed, @125us:

6000 - maxes out CPU at 100%
6500 - CPU Nice and low.

Speeding up the clock as described in the code above makes the 6000 board run nicely at 125us. 

Am I reading the code wrong?  Does the SPI clock not get speed up after initialization?

borisbstyle, 2016-04-15:
yup....cyclone has MPU6000 and totally no problems. I also had another prototype boards without issues.

CC3D always had some issues if you ask me. Even naze32 outperforms it with i2c!

kc10kevin, 2016-04-15:
Ok.  Last comment/question, then I'll drop it.  

Still doesn't explain why a board with the exact same design with just the MPU different shows such different results.  Both the 6000 and 6500 are great MPUs with not that much difference between them.

Question still not answered.  Is the code correct?  Does the SPI clock stay "slow" after initialization?  Is it meant to stay that way?  If not, then maybe that's the reason the CC3D under-performed compared to the Naze on I2C.  Maybe the CC3D was being choked and couldn't live up to its potential?

Now I'm done.

lkaino, 2016-04-15:
I checked the code and to me it also seems that the 18MHz speed is not set back.

The only functions where speed is changed are: mpu6000SpiGyroInit(), mpu6000AccAndGyroInit() and mpu6000SpiDetect(). Out of these three only the mpu6000AccAndGyroInit() sets the speed back to 18MHz. mpu6000AccAndGyroInit() is never the last function (out of these three) called so it seems that the speed is left at 0,5 MHz.

borisbstyle, 2016-04-15:
@kc10kevin 
I cannot explain it as the MPU6000 driver code has been unchanged since I started with this. 

I measured the read speed on cyclone and it takes 1-2us before it reads the gyro samples. That would not be possible on 0,5mhz So based on that I would say that the drivers works well.
If you find any improvements please let me know..

kc10kevin, 2016-04-15:
Ok.  I get where you are coming from based on your read speed test. 

 I guess the only improvement would be to speed the SPI clock back up in the MPU6000 driver code as I suggest in the first post.  It would only help to improve performance and reduce system/CPU load.  I've tested several different clock speeds and just by increasing the clock speed back up reduces load on the CPU.   Shown at all cycle times to reduce the load.  

I truly believe that when the code was written ages ago, there was an error.  To see that, you just have to look at the code contained in mpu6000SpiGyroInit.  It calls mpu6000AccAndGyroInit which slows the clock for the "slow" register writes, then speeds it back up to 18mHz.  When it returns to mpu6000SpiGyroInit, it is immediately slowed down again.  

Just a thought.  Thanks for taking the time to look into this issue and answer my questions.  

Offer still stands for me to send you a couple of the boards I designed.  If not, no worries.

borisbstyle, 2016-04-15:
@kc10kevin 
So you actually tested it and saw improvement there......than the code should be changed. I still don't understand why it would be fine on Cyclone and not fine on CC3D? 

Feel free to propose a PR into current code. I will test it too on othe boards and see if there is an improvement. Are you also able to test F1 CC3D and see what it does there?

kc10kevin, 2016-04-15:
@lkaino 

Thank you for verifying that I was reading the code correctly.  I got my BS is CS back when punch cards were in use.  Been a while.  :-)

kc10kevin, 2016-04-15:
@borisbstyle 

i did.  I started with the 6500 driver code since the board ran fine at 125us.  Looked at the 6000 code and found the "discrepancy" between the 6000 and 6500 code.  

I don't understand either, why the Cyclone runs just fine but not the CC3DF3.  I don't have a Cyclone to test it out on but I do have a few CC3Ds laying around.  Both the F1 and F3 versions.   I'll test the CC3D, CC3DF3 and my board I designed.   

Give me a couple days and I'll post up results here.  I don't have near the test equipment you have, so this will be off the CPU load only.  Sorry about that.

kc10kevin, 2016-04-15:
I'm pretty sure I figured it out why the Cyclone does not have problems and the CC3DF3 and my board do.   

I'm pretty confident that the code needs to be changed.   Here is why:

The Cyclone uses the Motolab hex.   In the Motolab target, both the MPU6000 and the M25P16 use SPI2.  I believe that the Motolab target is used by the cyclone and the Tornado (MPU6050) There is no M25P16 on the Cyclone from what I could see but is still  tried to be initialized by the firmware.

So, During startup, the gyro is initialized (MPU6000 via SPI2) first.   The SPI Clock is slowed down for the register writes, sped up to 18mHz, then slowed right back down and not sped up again.   Later in initialization, the M25P16 init ( m25p16_init() )is called which does the following:

bool m25p16_init()
{
    //Maximum speed for standard READ command is 20mHz, other commands tolerate 25mHz
    spiSetDivisor(M25P16_SPI_INSTANCE, SPI_18MHZ_CLOCK_DIVIDER);

    return m25p16_readIdentification();
}

So, SPI2 clock is sped back up to 18mHz, which is used by the MPU6000, not the flash.  That is why I believe the Cyclone works great at 125us but not the CC3DF3 or my board.  The clock never gets sped up since the M25P16 on the other boards since it is on different SPI instances.  

I cannot test this theory since I do not have a Cyclone board.  I think it can be tested very easily but building a target hex for the cyclone with the M25P16 instance (not needed anyway) commented out.   Any chance someone can do this?   Comment it out, build, flash and try to run at 125us.  

I think there is only the Cyclone and CC3D with the MPU6000 out there (besides my board).  Others on SPI use the 6500 or 9250 from my research.

Let me know if there is any confusion in the above.

Thanks.

borisbstyle, 2016-04-15:
Well that sounds reasonable indeed.

Can you do a PR with FIX for CC3D?

kc10kevin, 2016-04-15:
Will do.  Just to be sure, I'll still test the CC3D first.

Give me a day or two to test, make the change and submit a PR.  Sounf good?

borisbstyle, 2016-04-15:
Yes thats fine. I will verify cyclone

borisbstyle, 2016-04-15:
Well it should be matter of just adding one line back to mpu6000SpiGyroInit? Right?

borisbstyle, 2016-04-15:
What do you think about just adding this? Please test it
https://github.com/borisbstyle/betaflight/pull/347

kc10kevin, 2016-04-15:
That looks right.   I added it to the end of the procedure, but that should work too.  I wasn't sure if the clock needed to be slowed for the dlpf register write.   It's been tested and flown with it added at the end ( see first post for where I put it in).   I'll test it with your PR tomorrow later this evening.

Thanks!!

elf128, 2016-04-15:
I would strongly recomend to connect osciloscope to SCLK line and see what
exactly is going on.
So far, I've seen couple of places in SPI code, when certain initialisation
is necessary, but never done right. Such as CS pin initialisation of
SpiFlash get initialised by lucky coincidence in SPI init code and it
doesn't work any longer if you have CS on 'non-standard' pin.
Because this code has been written under assumption that you always has
only one SPI-slave device, I would not be surprised if some other
initialisation change SPI divider and everithing is running nice and
smooth, until you disable something in configuration.

dropax, 2016-04-15:
I think it's safe to say the fix (#347) is working and can be merged.
It saves around 100us for each gyro and acc readout. :)

before
Task list:
0 - (SYSTEM) max: 4us, avg: 0us, rate: 9hz, total: 2ms
1 - (GYRO/PID) max: 534us, avg: 271us, rate: 993hz, total: 38sec
2 - (ATTITUDE) max: 334us, avg: 299us, rate: 99hz, total: 4sec
3 - (ACCEL) max: 402us, avg: 133us, rate: 333hz, total: 7sec
4 - (SERIAL) max: 27263us, avg: 2us, rate: 99hz, total: 300ms
5 - (BEEPER) max: 8us, avg: 0us, rate: 99hz, total: 20ms
7 - (RX) max: 88us, avg: 56us, rate: 49hz, total: 561ms

with fix
Task list:
0 - (SYSTEM) max: 4us, avg: 0us, rate: 9hz, total: 1ms
1 - (GYRO/PID) max: 418us, avg: 166us, rate: 1002hz, total: 17sec
2 - (ATTITUDE) max: 315us, avg: 277us, rate: 99hz, total: 3sec
3 - (ACCEL) max: 287us, avg: 33us, rate: 332hz, total: 1sec
4 - (SERIAL) max: 26426us, avg: 2us, rate: 99hz, total: 132ms
5 - (BEEPER) max: 8us, avg: 0us, rate: 99hz, total: 15ms
7 - (RX) max: 89us, avg: 58us, rate: 49hz, total: 408ms

Good catch Kevin!

kc10kevin, 2016-04-16:
@dropax 

Thanks!  Glad to be able to let the CC3D breath a little bit.  I thinks it's been stifled for along time.  

Ready to close this out?

closed 2016-04-18 by borisbstyle