betaflight/betaflight #345: Spracingf3 evo

dal75, 2016-04-14:
Flashing the latest hex gets the board stuck in dfu mode. Then it's no longer recognised by the computer and can no longer access the gui. Only way back is by reflashing Cleanflight then it works normally again.

ReneHe, 2016-04-14:
Hi look too my close Issue. Flashing only with Dfuse 3.0.2

kc10kevin, 2016-04-14:
Boris posted this up in his Betaflight blog:

> I will remove evo hex from current release. I pretty much ported it from cleanflight without testing as I dont have this board and I need it to be able to test. I think it is missing something small like 1 line of code as some preprocessing defines had different names. All spracing targets are somehow special it seems.
In the meanwhile those users can just use cleanflight. Just use lower looptimes and it will be fine.

http://www.rcgroups.com/forums/showpost.php?p=34521729&postcount=25541

Arakon, 2016-04-14:
The actual error code it flashed with the BF hex is "ACC missing", so there's likely something wrong with the SPI init of the MPU.

borisbstyle, 2016-04-15:
@Arakon 
I have no idea. I cant debug without hardware. Maybe someone else who has the board will chimme in and help.

borisbstyle, 2016-04-15:
I think spracingf3evo misses some IO definitions like SPI CS, MISO pins etc....

borisbstyle, 2016-04-15:
Can anyone try this bin or hex?
https://dl.dropboxusercontent.com/u/31537757/SPRACINGF3EVO.zip

Arakon, 2016-04-15:
Will test as soon as I get home.

borisbstyle, 2016-04-15:
@Arakon 
I dont think it will fix anything by second thought, but try it

Arakon, 2016-04-15:
Same issue with that hex, 2 long error flashes.

Arakon, 2016-04-15:
@Boris: I kind of got it to work. 
#define USE_ACC_MPU6500 and #define USE_GYRO_MPU6500 had to be added in target.h. This way, the FC actually runs. Only issue now is that the Mag is going completely nuts. But at least it seems to boot.

Arakon, 2016-04-16:
Seems the "proper" way to go about it is to adjust the mpu6500 initialisation and setup from the initialisation.c from betaflight. The sections differ some between them.
Most/all of it is part of commit 8704717975b6d1c97fe6cffabe39d8b846488a63.
After these changes, the board seems to run okay, even without adding the target.h changes.. mag still goes nuts, though.

borisbstyle, 2016-04-16:
I believe there are some general issues with that mag, but I guess nobody
uses it anyway :)

Arakon, 2016-04-16:
Now the only issue I see is that the looptime jumps to 560-570 every few seconds. Could that be related to the fix?

borisbstyle, 2016-04-16:
I can only imagine baro doing that. disable it if it is not already

closed 2016-04-18 by borisbstyle