betaflight/betaflight #352: "dump all" switches to wrong rateprofile

rav-rav, 2016-04-16:
Let's say I'm using rateprofile 0.
"dump all" will dump all profiles and all rateprofiles inside them.
After closing CLI however, the rateprofile will be switched to 3.

I have to enter CLI again and type "rateprofile 0" to fix it. Or type "save" after dump all to avoid the change.

This command should not have any effect on the currently selected profiles.

Arakon, 2016-04-17:
I can confirm this, was confused recently why suddenly my rates were all off. I think what happens is that the command switches profiles to 0, 1 and 2 to display them, then doesn't switch back to 0 or whatever was currently active.

borisbstyle, 2016-04-17:
That is normal as it needs to switch profiles to be able to dump it.
I will add a line what remembers the old profile and switch back after dump
next time.

mikeller, 2016-05-10:
Had a look at this, and this seems to work fine now.

RipperDrone, 2016-06-06:
@borisbstyle Seems still flawed, I tried the following sequence in CLI:

rateprofile 0 - save - dump all - rateprofile  --> still showing rateprofile 0

Then disconnect / reconnect in Cleanflight GUI --> CLI rateprofile shows rateprofile 2 (bug?)

mikeller, 2016-06-06:
@RipperDrone: There still was a bug with the rateprofiles not being properly restored *per profile* as they should. Fixed this in #453.

RipperDrone, 2016-06-07:
Great, looks like it should be fixed then.

KiteAnton, 2016-06-23:
Fixed in #453

closed 2016-06-23 by KiteAnton