betaflight/betaflight #355: Update MWVersion with each FC version update

Hwurzburg, 2016-04-17:
I just noticed that MWVersion (used in an MSP response) has not been updated with the FC version....MWOSD uses this to display in its power up screen....would be nice if it could track the FC version for us OSD users( I use it even on my tiny Atom)...MWOSD has been keeping up with  BF,CF, and betaflight MSP changes, so it would be nice to have its startup screen accurately reflect the current FC rev...

will post an issue in CF github also...

borisbstyle, 2016-04-17:
OSD just needs to read proper version number. It is currently reading MW
version instead of firmware version.

Hwurzburg, 2016-04-18:
thanks...looking at code I see both reported....will raise issue with MWOSD firmware guys

closed 2016-04-18 by Hwurzburg