betaflight/betaflight #359: LED Feature on Naze32 rev 6 doesn't work

ps915, 2016-04-18:
Cant get WS2812 leds working on Betaflight.
I tried these versions:  2.6.1, 2.6.0, 2.5.4, 2.5.3

On one of my other quads i have 2.5.3 on a flip32 and it works with the same settings. So there must be something wrong with the naze32 rev 6

Settings on Naze32 rev 6:

1. RX_SERIAL
SBUS
PORT enabled Serial RX
= does not work

2. RX_PPM
SPEKTRUM1024
= does not work

On the latest Cleanflight i works.
LEDs are connected to CH5.
I just flashed stock Betaflight and then configured RX and LEDS, nothing else.

[image]
[image]

KiteAnton, 2016-04-19:
Check your settings and hardware connections again. I just tried this on a Naze rev6 with latest betaflight and it seems to work as expected. Sometimes you need to reboot the board after LED settings changes to make them work.  

Can you post your dump ?

KiteAnton, 2016-06-23:
Old issue, closing since seems unrelated to FW issues.

closed 2016-06-23 by KiteAnton