betaflight/betaflight #361: CC3D Atom and PPM input not working

acidg, 2016-04-20:
I have flashed version 2.6.1 to my CC3D Atom and am using a FrSky DR4-II in PPM mode. However, the input is not recognized. 
I used pin 3 and 6 of the receiver port as well as pin 6 of the servo port.

The board is working with port 3 of the receiver port and Cleanflight 1.10.0

Arakon, 2016-04-20:
If you are counting the original PPM pin as 3 (it's actually 1, VCC and GND don't count), then you need to use pin 8, not 6. Last pin of the receiver port.

acidg, 2016-04-21:
Sorry, I meant pin 1 and 6. I tried both of the pins, but the signal is not recognized.

renejsanchez, 2016-04-24:
Last pin of the receiver port, pin 8, worked for me.

acidg, 2016-05-09:
Pin 6 (the last one) doesn't work for me. @renejsanchez are you also using a D4R-II receiver from FrSky in RX_PPM mode?

Headi1986, 2016-05-12:
I have also problems with CC3D ATOM,D4R-II and PPM on Pin 8 with oneshot and without oneshot, it doesnt work. Betaflight 2.3.4 is working fine with PPM on Pin 3, but releases after that are not working.

borisbstyle, 2016-05-12:
That pin has been moved due to timer issues and loss control on PPM.

Please read release notes after 2.3.4

Headi1986, 2016-05-13:
Now i have tested all pins from the Receiver Port, only Pin 3 works with version 2.3.4. I think there was a problem on Pin 8 of my CC3D Atom black pcb. My second CC3D Atom, also black pcb does not work too, only Pin 3 and PPM works. With my normal CC3D (black pcb) i have no issues with PPM + oneshot and Pin 8.

borisbstyle, 2016-05-24:
Cool. I will close this issue to prevent cluttering

closed 2016-05-24 by borisbstyle

renejsanchez, 2016-05-29:
I'm using an orangerx r410x in rx_ppm mode.