betaflight/betaflight #362: SPRACINGF3EVO ppm dont work

jcaamnes, 2016-04-21:
PPM on the SP Racing F3 EVO does not work in betaflight. I had to flash back to cleanflight to make it work :(

Boappp, 2016-04-22:
Same problem, and I work on cleanflight firmware.

schmiernippel, 2016-04-30:
Same problem her with cppm ! With cleanflight everything works !

Arakon, 2016-04-30:
This board does something weird with PPM: The PPM pin is actually re-assigned to RX on UART2. I think to get all functions working, hydras support may be needed.

schmiernippel, 2016-05-03:
Any news when there will be a fix ? 

greets

borisbstyle, 2016-05-04:
@schmiernippel 
First problem has to be found :D

schmiernippel, 2016-05-04:
Ok, hopefully there will be a solution ! :-)

hydra, 2016-05-04:
@borisbstyle check the support for out of order ISR's in the PPM source in cleanflight.

Arakon, 2016-05-04:
PPM Fix that was commited a few ago seems to work fine, just tested with a D4R-II.

schmiernippel, 2016-05-05:
Where can i download the fixed firmware ?

jcaamnes, 2016-05-05:
@schmiernippel Can you provide a link to the fixed firmware?

schmiernippel, 2016-05-05:
http://andwho.sytes.net:8080/job/BorisB_BetaFlight/304/ 

Click on "expand all"

schmiernippel, 2016-05-05:
It works for me ! Thanks ! :-)

closed 2016-05-05 by borisbstyle