betaflight/betaflight #375: Spracing EVO  Not arming on switch

Davidsg33k, 2016-05-03:
on 2.6.1 i can not seem to get the ship to arm.
checked all the normal stuff.

Any ideas are welcome.

schmiernippel, 2016-05-03:
Does it arm with stick movement ?

borisbstyle, 2016-05-03:
check min_check value

lunddahl, 2016-05-03:
My 3 SP Racing EVO boards arms just fine, but was support not pulled for this Board for 2.6.1, i have to use newer Jenkins builds on those boards.

http://andwho.sytes.net:8080/job/BorisB_BetaFlight/

This is the first one to include the fix:
http://andwho.sytes.net:8080/job/BorisB_BetaFlight/289/
https://github.com/borisbstyle/betaflight/commit/4186ceaf81ec7bd14374a7445c96ab29acd9c109

Davidsg33k, 2016-05-03:
Tried the minimum command value change,  no good..    Borisbstyle if i can supply more information, let me know what you need, will supply.

and lunddahl will try the fw you linked to.   thank you all, looks like there has been issues with the fw of this board so far.    "would like to clean it up."

Davidsg33k, 2016-05-03:
also, this only happens with the betaflight version,  but cleanflight will not do blheli passthrough, and thats handy.

Davidsg33k, 2016-05-03:
[image]

[image]

[image]

Davidsg33k, 2016-05-03:
[image]

[image]

flounderscore, 2016-05-03:
You can only arm at minimum throttle. What's the "ms" throttle reading at
low throttle stick?

Davidsg33k, 2016-05-03:
[image]

flounderscore, 2016-05-03:
There you go, that value is too high. I forgot what the maximum acceptable
value is (google is your friend), but I typically set my throttle to the
1000 ms - 2000 ms range (as all my other channels).

Davidsg33k, 2016-05-03:
so tell me why it works in clean flight,   but not betaflight with the same settings?   

i have adjusted that setting with no change.......    flounderscore   so fill me in..   im not new to this.

flounderscore, 2016-05-03:
https://github.com/borisbstyle/betaflight/releases/tag/v2.5.1

 "Default min_check lowered to 1040 (make sure your radio can go below 1040
to be able to arm)"

Davidsg33k, 2016-05-03:
Thank you,   this has fixed my issue.    you all rock..  

1040 check,  go figure...

closed 2016-05-03 by Davidsg33k