betaflight/betaflight #383: Per cell telemetry emulating FLVSSS

MasterZap, 2016-05-06:
See Cleanflight pull request https://github.com/cleanflight/cleanflight/pull/1927/files

And no, the "cell average in a4" already merged does NOT satisfy.

The usecase is: Have ONE model on your radio, have multiple quads, some with a real FLVSSS sesor (that sends you actual cell voltages), some without... and being able to use same per-cell voltage alarms on either.

Now, it¨s a major mess. with BetaFlight telemetry over SPort, voltage comes in at a different route than when you have an actual FLVSSS, plus, the precision is not enough. The actual cell telemetry package has much higher precision.

MasterZap, 2016-05-06:
(any argument in the form "but emulated cell telemetry is dangerous, what if a single cell is low" will be summarily ignored, thankyou :) )

borisbstyle, 2016-05-09:
What another different cell reporting?

MasterZap, 2016-05-09:
No, just one done correctly (on the proper cell data channel) :)

It's really a minor change, replying to the right cell data packet in the cell data packet format. There is already code for it, but someone decided (wrongly, IMHO) to send this over "A4" instead, which makes zero sense to me.

QuadroBro, 2016-05-11:
You sure  are uppity for a dude getting free shit...

MasterZap, 2016-05-16:
I'm just frustrated when the "right" code was/is there and someone made an alternate "wrong" implementation and that was picked instead....

KiteAnton, 2016-06-17:
@MasterZap can you make this into a PR ?

MasterZap, 2016-06-21:
I solved this instead w. a script on the radio side, so I personally do not need this any more. It would still be a great feature, though.

closed 2016-06-21 by KiteAnton