betaflight/betaflight #385: KISS 24A onewire telemetry

IB1387, 2016-05-07:
Hi all,

adding support for the KISS's 24Amp onewire telemetry would be really nice. Just an idea, keep up the good work!

-1387

borisbstyle, 2016-05-09:
@IB1387 
I dont think this likely be added.IMO it would be better to add this directly onto some OSD.

Linjieqiang, 2016-05-14:
I want to know how " the Kiss ESC onewire telemetry" to implement? Serial ports?Or else communicate way?

FabianViermetz, 2016-05-16:
http://ultraesc.de/downloads/KISS_telemetry_protocol.pdf this is the documentation for the protocoll, semms that you just need one serial port plus the pwm line. i am currently working on an implementation especially for the kissfc version of betaflight.

IB1387, 2016-05-16:
That would be so awesome. I just purchased a KissFC  in hope that if I don't live the firmware that Betaflight could be ready until then. How long will it approx. take to finish the betaflight Port für the KissFC?

FabianViermetz, 2016-05-16:
be careful with that firmware, it might be stable in most cases but i had serious security issues with their released beta (motors started when calibrating the acc (dont think thats part of the calibration)) will try out the bf port for the kissfc today and start with the telemetry 

@borisbstyle the idea with the osd is probably not possible since most osd have just an ATMega328 with one serial port and telemetry comes with 115200 baud so an softwareserial is not possible ether

borisbstyle, 2016-05-19:
I am not really doing much with serial stuff. Maybe someone else will ever look into this with that specialty.
Also someone else can port betaflight to KISS FC. There is already draft. For now I dont see any urge to do it as KISS hardware is designed for their own firmware. Some features would be tricky to introduce. For example passthrough would never work because of the buffers.

KiteAnton, 2016-06-17:
Old issue and also stated unlikely to be added. I close this one, if someone wants to make a PR for the KISS onewire telemetry then do so and it can be reviewed/discussed further then.

closed 2016-06-17 by KiteAnton