betaflight/betaflight #387: Restore from 'dump all' sets profile to 1

mikeller, 2016-05-08:
After restoring an FC configuration with the data collected by a `dump all` CLI command, profile 1 is left as the active profile.
I think before the profile / rateprofile rework, the profile was switched back to 0 by having a `profile 0` command at the end of the dump. Maybe this should be re-added, as it reduces the risk of flying with an unconfigured profile for users with only one profile set up.

Also, after restoring from `dump all`, `yaw_iterm_reset_degrees` is left at the default value of 50 for profile 0, despite `set yaw_iterm_reset_degrees = 300` being part of the dump output for profile 0. This seems to be somehow related to profile switching during restore, as manually switching to profile 0 and issuing `set yaw_iterm_reset_degrees = 300` sets the value just fine.

mikeller, 2016-05-09:
Holy cow.

Just tested restore on an FC, and had a look at what happened in the CLI window.

This part of the output of `dump all`:

```
set tpa_breakpoint = 1500

# profile
profile 1
############################# PROFILE VALUES ####################################
profile 1

set yaw_p_limit = 400
set dterm_lowpass = 80
set dynamic_pterm = ON
set iterm_always_reset = OFF
set iterm_reset_degrees = 200
set yaw_iterm_reset_degrees = 50
set yaw_lowpass = 70
set pid_controller = MWREWRITE
```

turned into this:

```
# set tpa_breakpoint = 1500
tpa_breakpoint set to 1500
# # profile

# profile 1
profile 1

# yaw_lowpass = 70
Unknown command, try 'help'
# set pid_controller = MWREWRITE
pid_controller set to MWREWRITE
```

Looks like with the current firmware it takes a lot more `#` to compensate for the delay introduced by the profile switch.

borisbstyle, 2016-05-09:
If it would switch back to 0 that will be confusing who fly on profile 1. I believe I tried to let it switch back to its original profile, but that apperantly needs some debugging to make it perfect

KiteAnton, 2016-05-09:
Yes it was discussed in #273 that the input buffer does not always cope with the input rate from cli when doing a dump. I made a [PR](https://github.com/cleanflight/cleanflight-configurator/pull/354) for this to cleanflight configurator some time ago but it seems that the development is slow .. I have the change running on my computer locally and it fixes the issue.

mikeller, 2016-05-09:
> If it would switch back to 0 that will be confusing who fly on profile 1.

I would assume that the number of people flying with profile 1 and not having configured profile 0 is much smaller than the number of people flying with profile 0 and not having configured profile 1, based on the fact that profile 0 is the default that is selected after a reflash.

> I believe I tried to let it switch back to its original profile

As in 'the profile that was selected at the time the dump command was issued'? That would be preferable to always switching back to 0. But in the absence of that, switching to 0 is probably better than leaving it on 1, as 'people flying on profile 1 and not having configured profile 0' is most likely a corner case.

mikeller, 2016-05-09:
#389 adds commands to restore the profile / rateprofile selection to the output of 'dump all'. This is the fix that @borisbstyle has suggested.

KiteAnton, 2016-06-17:
This seems to be fixed so I close the issue.

closed 2016-06-17 by KiteAnton