betaflight/betaflight #393: V2.7 Default looptime on F1 targets possibly too low

Bahecuk, 2016-05-12:
After clean install of Betaflight 2.7 with chip erase looptime on my CC3D was set to 500(2KHz) with accelerometer enabled by default, 3D model of the quad moves in cleanflight configurator, but problem is that my looptime is all over the place with jumps into ~800 range even with UART connection.
Is it supposed to be that way or just a wrong default value?
One more person on rcgroups reported the same issue on naze32 board.

Bahecuk, 2016-05-12:
This is what "#status" shows with 2.7 default settings (500 looptime and accel enabled):
```
# status
System Uptime: 24 seconds, Voltage: 0 * 0.1V (3S battery - OK), CPU:6%
CPU Clock=72MHz, GYRO=MPU6000, ACC=MPU6000
Cycle Time: 757, I2C Errors: 0, config size: 1276
```

borisbstyle, 2016-05-12:
just set it to whatever works if thats 1000.

500 is great for F3 boards and thats the current mainstream.

Bahecuk, 2016-05-12:
It's not a problem for me, I set 375 looptime and it works fine, at least on the bench.
I'm mostly worried about new guys who don't check if looptime is stable with current default settings and will attempt to fly with 2KHz loop frequency on F1 board with accel enabled and looptime badly fluctuating. Is it possible to change defaults only on F1 targets? Or at least a note on future releases like: "F1 target users need to change looptime to 1000us, or disable accel by manually choosing lower looptime"

borisbstyle, 2016-05-12:
500 should work I think if you dont have many features enabled.

On naze many folks disable acc anyway :)

I will add a note in release notes

Bahecuk, 2016-05-12:
cool, thank you, Boris)

closed 2016-05-12 by Bahecuk