betaflight/betaflight #395: 本人於13日更新BF程式由2.6.1更新為2.7.0飛控版為SPF3.發生電機與電變燒毀.後來查CLI發現增加配置快速PWM協議，增加了新的方式（CLI：fast_pwm_protocol（ONESHOT125，ONESHOT42，連拍））.BF2.6.1Oneshot42實施（當單穩功能啟用類型“設置use_oneshot42 = OFF”） 連拍實施（當oneshot125功能啟用類型“設置use_multishot = OFF”）.使用上都無問題.但是更新為2.7.0程式後就有問題.     希望大家要注意.應該2.7.0程式裡有BUG.希望英文夠好的飛友向BF反應.在此不希望有下個受害者.我已經損失慘重KISS18A與銀燕第二代2204都燒燬.再一次提醒大家小心.我不知網上是否有飛友下載.是否跟我類似問題.

E120103016, 2016-05-14:
本人於13日更新BF程式由2.6.1更新為2.7.0飛控版為SPF3.發生電機與電變燒毀.後來查CLI發現增加配置快速PWM協議，增加了新的方式（CLI：fast_pwm_protocol（ONESHOT125，ONESHOT42，連拍））.BF2.6.1Oneshot42實施（當單穩功能啟用類型“設置use_oneshot42 = OFF”）
連拍實施（當oneshot125功能啟用類型“設置use_multishot = OFF”）.使用上都無問題.但是更新為2.7.0程式後就有問題.
    希望大家要注意.應該2.7.0程式裡有BUG.希望英文夠好的飛友向BF反應.在此不希望有下個受害者.我已經損失慘重KISS18A與銀燕第二代2204都燒燬.再一次提醒大家小心.我不知網上是否有飛友下載.是否跟我類似問題.

digitalentity, 2016-05-14:
I doubt @borisbstyle or any other dev here reads Chinese and therefore won't be able to help you. Please consider describing your issue in English.

nathantsoi, 2016-05-14:
what version of blheli are you using? i think you might need a version that disables pwm...

also, what was your looptime?

also, I agree with @digitalentity, english would be better :)

E120103016, 2016-05-14:
1.ESC為KISS 18A沒有用到blheli.
2.LOOPTIME為BF2.7.0原設定.

borisbstyle, 2016-05-19:
Whaaaaaat? lol

new command is
set fast_pwm_protocol = oneshot42

closed 2016-05-19 by borisbstyle