betaflight/betaflight #396: Issue with LUXFLOAT in BF 2.7.0

fl0rene, 2016-05-15:
Yesterday I switched to BF 2.7.0 on my NAZE rev.6! But when switching one of my PID-profiles to LUXFLOAT all my sensors are gone!!
No BARO, no ACC, no MAG and no GYRO is recognised by Cleanflight-configurator!!!
After I pressed 'reset-settings' and reboot of FC all sensors are recognised again!!!

Switched back to BF 2.6.1 and the issue is gone!!!
Have I done something wrong or is this a bug of 2.7.0???

JurikRistjouw, 2016-05-15:
It could be default behviour like when looptime is set to 500. This will automatically disengage these sensors by removing acc task from the loop. like with the cli_command. You could try to reenable it with cli command.

fl0rene, 2016-05-15:
Thanks Jurik!

You are right!!!
Solved all problems with luxfloat in BF 2.7.0.

Thanks again..

borisbstyle, 2016-05-16:
Will be fixed in 2.7.1 release

closed 2016-05-16 by borisbstyle