betaflight/betaflight #397: vibration

Ajare77, 2016-05-15:
I have a big vibration in stabilization, I tried default PIDs (i Rewrite Luxfloat) settings from the previous version 2.6.1 (not ideal but usable), looptime 1000 setup naze32rev5, 2804/2300, 12A (no-shooted), 5040 , frame260
I also tried the CLI parameters gyro_lowpass and dterm_lowpass, also without success ...

borisbstyle, 2016-05-19:
Check your magnets.

Most vibrations are still mechanical and not software related ;)

closed 2016-05-19 by borisbstyle

Ajare77, 2016-05-19:
> Komu: borisbstyle/betaflight <betaflight@noreply.github.com>
>
>
>
> "

Magnets are ok, but just change my bearings ... I wonder what will be the 
effect :-)

Also waiting for shipment Cobra motors, controllers LittleBee20A-PRO and 
control unit SP Racing F3 Acro, so there will be plenty of space for 
additional debugging ;-)

Thank You for your valuable time and information, have a great day.

Roman

By the way...You're doing really luxurious work, I'm a big fan of yours

---------- Původní zpráva ----------

(https://github.com/borisbstyle/betaflight/issues/397#issuecomment-220272566)

"=

Ajare77, 2016-05-19:
With version 2.7.1, the situation is slightly better, but is still wobbling...

borisbstyle, 2016-05-24:
Test your motors in motor output. verify if those are fine

Ajare77, 2016-05-24:
Thanks for the advice, the engines are not in the best condition, however i 
always trying balancing. When arrive new, soon we'll see if the problem is 
in motors ;-)

---------- Původní zpráva ----------

Datum: 24. 5. 2016 14:04:43

"

(https://github.com/borisbstyle/betaflight/issues/397#issuecomment-221248084)

"=