betaflight/betaflight #399: Heading Drift

BoyDrone, 2016-05-16:
Hi Boris, I’m new to this, so forgive me if I make an error in this issue report or if you require additional data. I haven’t had too much flight time, but just bench testing quite a bit before ramping up flight time.

I’ve noticed some heading drift while connected to my quad via USB. I noticed it on the latest release, but maybe because it was more prominent and I was looking for anomalies. I recorded the drift with a timer, nothing fancy, just watching the values and logging. Someone had suggested reverting to an older stable version, and it seemed to drop significantly.

Some notes on my setup:
Naze32 Rev5 board
Connected via USB cable, without power (no radio inputs)

Software version: 2.1.6 Drift: +1 deg / 37 seconds
Software version: 2.6.1 Drift: +1 deg / 13 seconds
Software version: 2.7    Drift: +1 deg / 15 seconds

Considering the variance rate, I think this is a software issue unfortunately…. Although if it was hardware, it could just be pronounced in software. I’ve noticed others with similar issues with older versions of cleanflight on other boards, but wasn’t sure if this was logged on betaflight.

I look forward to seeing feedback on this, even if it refers me to another thread. Thanks for all your hard work!

flounderscore, 2016-05-16:
Just a guess, is your looptime 4 times greater on 2.1.6?

BoyDrone, 2016-05-17:
I hadn't changed anything from the default values, just flashed as monitored heading.

borisbstyle, 2016-05-19:
Try 2.7.1 and lower lowpass.

Vibrations can cause more drift on noisy setups. 2.7 has less agressive filtering.

closed 2016-05-19 by borisbstyle

BoyDrone, 2016-05-19:
Thanks Boris. I'll try this out tonight for sure.